modelica-3rdparty/Modelica_DeviceDrivers #49: Rebuild binaries in master

tbeu, 2015-02-06:
Bernhard, can you please rebuild and update the binaries of the master branch. Thank you.

bernhard-thiele, 2015-02-10:
I didn't rebuild the space mouse and CAN part, since I guess it was ITI_MDD.dll that the needed an update due to changes in the real-time synchronization code. Hope that closes the issue.

closed 2015-02-10 by bernhard-thiele

tbeu, 2015-02-10:
Can you please use again VS2010. Otherwise it will not work on WinXP. Thank you.

bernhard-thiele, 2015-02-13:
Any way to set options to VS2013 to make it work?

My notebook SSD is a bit short on disk space ...

tbeu, 2015-02-13:
I see. I try to do it on my own though I have no Cmake experience at all. I'll push to my fork then,

tbeu, 2015-02-16:
Do you consider to merge the VS2010 binaries?

bernhard-thiele, 2015-02-17:
I've been away for a few days. Just merged it. Thanks for providing the VS2010 builds.

tbeu, 2015-02-17:
Thanks. This makes it easier for me if your master and my forked master are in sync.

commit 57a9e3e3 (2015-02-10): Fixes #49