modelica-3rdparty/Modelica_DeviceDrivers #63: Inconsitency of inclusion of ModelicaUtilities.h

tbeu, 2015-02-25:
In the Library include files we find includes like this
```c
#include "ModelicaUtilities.h"
#include "../src/include/CompatibilityDefs.h"
```
in cas of MSVC. But ModelicaUtilities.h is located in same directory as CompatibilityDefs.h is. Thus ModelicaUtilities.h is not found without explicitly adding src/include to the include path. Thus I propose to get either rid of the extra src/include directory by adding all files to the main Include directory. Or alternetively the include directives should be made consistent.

bernhard-thiele, 2015-03-04:
The idea with `ModelicaUtilities.h` was following:

- If simulated using a tool (in this case Dymola) it should take the `ModelicaUtilities.h` that ships with the tool
- if using the file for C-only unit tests without Modelica it should use the file in `../src/include/` which is achieved by adding that path to the include paths when compiling for C-only unit tests.

I was not sure to that time  whether one may always assume that they are equal  and wanted to be on the "safe" side. However, I guess they should be and agree that it should be moved as suggested.

closed 2015-03-04 by bernhard-thiele

tbeu, 2015-03-04:
Yes, ModelicaUtlities.h should be tool-independent.