modelica-3rdparty/Modelica_DeviceDrivers #68: Windows compilation

tbeu, 2015-03-05:
Windows compilation is restricted to MSVC. It should be straight forward to relax Windows compilation to also support MinGW and Cygwin.

tbeu, 2015-03-24:
Should be fixed: Welcome OpenModelica (on Win)!

What is left is
* adaption of the cmake scripts to also consider Mingw/Cygwin
* adaption of the cmake scripts to consider the new test for serial port communication
* add Library={winmm} to keyboard and joystick functions
@bernhard-thiele Are you willing to do this? If not, just close this issue. Thank you!

bernhard-thiele, 2015-03-24:
Yes, I will do this. It is on my list. It just has low-priority as long as the communication/packaging blocks don't work in OM. Getting them to work means changing some parts of the compiler. I might work on that if I have some spare time for it, but maybe someone else will fix it for OM.

tbeu, 2015-03-24:
You could start with input devices (that are independent of packager).

tbeu, 2015-10-08:
[x] adaption of the cmake scripts to consider the new test for serial port communication is done by 4a98e21d1cc6db16c84b91c8b9f96e946e9f657a

VanessaSingh, 2017-06-15:
Could you please list out the steps on how to compile the Blink example in Embedded Targets(in Serial Communication) on windows as it is giving  'unrecognized command line option "-std=c11"'
on both MinGW and MSVC.

sjoelund, 2017-06-15:
Which gcc version is in your avr-gcc? The latest Windows release of 8-bit avr-gcc contains gcc 4.9, which does support c11. If you installed the 32-bit avr-gcc, it seems that uses gcc 4.4 which was released in 2009 and indeed does not support c11.

VanessaSingh, 2017-06-15:
Yeah, that was the issue as it works fine with gcc 4.9. Thanks for the help!

tbeu, 2020-01-01:
All done.

closed 2020-01-01 by tbeu