modelica-3rdparty/Modelica_DeviceDrivers #69: TestSerialPackager_UDP: Initial conditions are not fully specified

tbeu, 2015-03-05:
```
Translation of Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_UDP:

The DAE has 102 scalar unknowns and 102 scalar equations.

The initial conditions for variables of type Real are not fully specified.
Dymola has selected default initial conditions.

Assuming fixed default start value for the discrete states:
getFloat.y[1](start = 0.0)
getFloat.y[2](start = 0.0)
getReal.y[1](start = 0.0)
getReal.y[2](start = 0.0)
getReal.y[3](start = 0.0)

The initial conditions for variables of type Integer are not fully specified.
Dymola has selected default initial conditions.

Assuming fixed default start value for the discrete states:
getInteger.y[1](start = 0)

Statistics

Original Model
Number of components: 33
Variables: 137
Parameters: 31 (31 scalars)
Unknowns: 106 (112 scalars)
Equations: 47
Nontrivial: 47

Translated Model
Constants: 30 scalars
Free parameters: 18 scalars
Parameter depending: 23 scalars
Time-varying variables: 26 scalars
Alias variables: 46 scalars
Assumed default initial conditions: 6
Number of mixed real/discrete systems of equations: 0
Sizes of linear systems of equations: { }
Sizes after manipulation of the linear systems: { }
Sizes of nonlinear systems of equations: { }
Sizes after manipulation of the nonlinear systems: { }
Number of numerical Jacobians: 0

Selected continuous time states

Finished

// experiment StopTime=5 StartTime=0

WARNING: 2 warnings were issued

Finished
```
Should this be fixed in the Get blocks or in the example model?

bernhard-thiele, 2015-03-05:
Yes, I was sloppy about that initial conditions. If it is fixed only in the example models, the user is forced to think about sensible initial values whenever using such a block. This is not a bad thing, but it is not as convenient as just drag-and-drop and "0" should be fine in many cases. 

So I have no strong opinion about it. I would tend to put it into the Get blocks, but if you think the Examples are the better place I'm also fine with it.

bernhard-thiele, 2015-03-08:
I guess the merge should close this issue.

closed 2015-03-08 by bernhard-thiele