modelica-3rdparty/Modelica_DeviceDrivers #77: Only clocked/sampled Communication

tbeu, 2015-04-14:
Currently the communication is cyclic only.

Proposal: This is taken from a trigger type implementation for CAN we once did and where more trigger options are available.
```mo
type trigger_type=enumeration(
  event_trig "Event",
  cyclic_trig "Cyclic",
  cyclic_event_trig "Cyclic + Event",
  initial_trig "At Start only",
  terminal_trig "At Stop only")"Use as trigger";
```

tbeu, 2015-04-14:
@dietmarw May be interesting for you that the syntax hiliting of the closing bracket is blue where it should be black.

dietmarw, 2015-04-14:
I'll ping @thorade as he knows more about what highlighter is currently used by GH.

thorade, 2015-04-14:
Modelica syntax highlighting was added as a reaction to this issue:
https://github.com/github/linguist/issues/2053
here is the corresponding pull request:
https://github.com/github/linguist/pull/2059

In that pull request, the repo@commit https://github.com/BorisChumichev/modelicaSublimeTextPackage/commit/d7e50e39c14a49153d3c17dfbd623258cf1a5d69 was chosen for syntax highlighting. The repo has since then seen some updates, so the pointer to the hash should be updated. But before doing that, you should file an issue that demonstartes your problem here:
https://github.com/BorisChumichev/modelicaSublimeTextPackage/issues
Upload some problematic Modelica code e.g. to Gist, then use Lightshow:
https://github-lightshow.herokuapp.com/
E.g. like this:
http://bit.ly/19e6Mv1

tbeu, 2015-04-14:
@thorade Feel free to do it. Thanks. :grinning:

tbeu, 2015-04-14:
@bernhard-thiele Do you think we can reuse PackageIn.trigger and PackageOut.trigger for alternative triggers? Can we use backwardTrigger only enable if trigger_type is cyclic? Is this alloed in a connector?

bernhard-thiele, 2015-04-14:
@tbeu Yes, I think so. I will try something once I have some time and a free dymola license. I will try to support an additional external Boolean trigger input signal which is conditional available if some Boolean parameter is set. The different triggering options could be realized by a distinctive "trigger" block. I think this would be a pretty flexible solution.

tbeu, 2015-04-15:
Good idea. I believe I can support you. Cylclic should be independet of Boolean trigger input to realize "Cyclic + Event" triggering.

bernhard-thiele, 2015-04-17:
@tbeu I just made a first attempt in a new feature branch. Example `Modelica_DeviceDrivers/Blocks/Examples/NEWTestExternalTriggerSerialPackager.mo` illustrates the idea.

Note that this is a backwards compatible change. What do you think about it?

tbeu, 2015-04-17:
Thanks for fast implementation. My thought:
 * Why did you choose to add the trigger port to the SerialPackage? Answering myself, one reason might be to do it only once at exactly one location in the package. But if I had done it I'd select the communication send/receive blocks for the addtional trigger port. In my opinion it logically belongs there. It is the possible to create a package and connect multiple send blocks (wih different triggers) to it.
 * There is no need to support the "Cyclic + Event"  triggering like we did in the past and I mentioned above. Best way of modeling is like mentioned above with one package and multiple communication blocks.
 * Imagine the case where two ore more packages need to be sent in a particular order (in order to realize some special communication protocol). This would require an extra Boolean outport returning the I/O success for the communication blocks for modeling chained communication processes.

Let me hear what you think about these comment. Thanks.

bernhard-thiele, 2015-04-18:
Ok, my thoughts: 

* I implemented the mechanism for the SerialPackager just as an example for discussion and planned to implement the same mechanism also for the various receive and send blocks. However, that alone is probably still not sufficient for the use case that you describe, since it wouldn't allow to connect several send blocks to one package. The triggering logic would need to be extended so that it can cope with several sinks or sources. For the sinks and sources one would need to be careful so that the same device can be shared. But maybe I'm thinking to complicated...
* The Boolean output is an interesting idea. But a simple I/O success might not be sufficient for modeling chained communication processes. We could either generate a new event that is capable of triggering the next blocks, or we could try to ensure that all blocks are synchronized at events and use an additional signal to establish data-flow dependencies that would ensure the correct ordering. What scenario to you have in mind?

tbeu, 2015-04-18:
* You are right: Multiple communication sinks are too complicated if possible at all. Maybe the extended trigger type is a solution.
 * I was thinking of custom protocols where packages need to be sent in a particular order, e.g, do a handshake first and then do sampled communication. It should be possible as custom text but maybe not with generic block components since different packages may be required for the different protocol steps.

bernhard-thiele, 2015-07-26:
Just created branch `externalTriggerFeature` in which I added an option to use an external trigger for triggering the read/write operations. This approach uses conditional connectors which adds another challenge to Modelica tools and people trying to understand the code ;-).

@tbeu Does the new code work in SimulationX and do you have any comments on it?

tbeu, 2015-07-27:
Can test it next week.

commit 3fcb779c (2015-08-11): Merge branch 'externalTriggerFeature'

Closes #77
closed 2015-08-11 by bernhard-thiele