modelica-3rdparty/Modelica_DeviceDrivers #78: TCP/IP support wanted

tbeu, 2015-04-20:
My branch add-TCP/IP-support at tbeu/Modelica_DeviceDrivers contains a not yet final implementation of a TCP/IP socket client (for Windows).
@bernhard-thiele Can you please test and estimate if you want to have this in the library? In Incubate?

bernhard-thiele, 2015-04-21:
Definitely, TCP/IP support would be nice to have! I wouldn't put it into Incubate, unless you view it is an experimental or incomplete approach.

I just would like to understand whether/why you want to diverge from the data source/sink principle used elsewhere in the library and have everything in one block. Or is this just due to the early prototype?

tbeu, 2015-04-21:
The reason is that there is only one instance of the external object for the TCP/IP client. This is required for the blocking ```recv``` of the client from the server. Thus it is not possible to separate input and output if I am not wrong.

closed 2015-04-21 by tbeu