modelica-3rdparty/Modelica_DeviceDrivers #83: Release plan

tbeu, 2015-05-20:
Any plans to release a new version? v1.3 was released one year ago and many fixes where added since then.

I also propose to switch to semantic versioning, i.e. 1.4.0 would be a proper name according to https://github.com/mojombo/semver.

bernhard-thiele, 2015-05-21:
Yes, we should have a new release soon. Thought of having a new release after implementing the external triggering proposal (in the externalTriggerProposal branch). Let's try for getting out a new release at the end of this month.

I like semantic versioning, the problem with it is currently the missing tool support. There is no sensible mapping to Modelica versioning annotations. It is not possible to express that any 1.4.x is fine if another library specified a dependency to 1.4.0. Think this was discussed in in some Modelica ticket some time ago. The current way for bugfix releases in Modelica is to increase the "build" number.

So, I think we shouldn't switch to semver until there is some better Modelica/tool support for it. I know that there are quite a few people who like semver in the Modelica Association, but I don't know if there is currently a concrete proposal or work going on in regards of this.

tbeu, 2015-05-21:
Actually I'd like to see the semver here. Currently it's just a third digit - nothing else. There is much work and good proposals on semver in MA. See e.g. https://trac.modelica.org/Modelica/browser/ModelicaDesign/trunk/MeetingMinutesMaterial/min84_2014_Lyon/min84.pdf

sjoelund, 2015-05-21:
I should note that OpenModelica has had "support" for semver (at least x.y.z) for many years. Adding a `uses(Modelica(version="1.4"))` loads 1.6. Uses 0.1 loads 3.2.1 (because there are no 0.x versions available). Uses 2.8 loads 3.2.1 because there is no 2.x where x >= 8. Uses 3.2.0.6 loads 3.2.1. And so on. It simply assumes it is possible to use a later version (because it does not know about conversion scripts).

I assume Dymola can do the same and see if there is a "no conversion needed" annotation. And then everyone would be happy.

bernhard-thiele, 2015-05-21:
OK. When let's have 1.4.0 as next version.

tbeu, 2015-05-21:
Fine.

closed 2015-05-21 by tbeu