modelica-3rdparty/Modelica_DeviceDrivers #84: New Release?

tbeu, 2015-07-24:
What is the schedule for a new version?

bernhard-thiele, 2015-07-24:
Should be released one month ago. I will try to get it done next week (need access to Dymola license to do it... )

tbeu, 2015-08-13:
In case we want to have some information on the new release in the upcoming Modelica newsletter we should meet the deadline September 4th. We can use this issue to discusss the content we want to place there. But time is hurrying...

bernhard-thiele, 2015-08-14:
I have played with some ideas from Mike and @dietmarw about how to organize the release of Modelica libraries that are discussed in this [blog](https://github.com/xogeny/impact/blob/master/resources/docs/modelica2015/paper/impact.md#impact-on-library-developers).

So, I created a toy "alpha" release at [https://github.com/modelica/Modelica_DeviceDrivers/releases](https://github.com/modelica/Modelica_DeviceDrivers/releases). It's not supposed to be visible to the public (unpublished draft), but I think @tbeu and @dietmarw can see it.

@dietmarw Is it already possible to point the link form [https://www.modelica.org/libraries](https://www.modelica.org/libraries) to that release page, instead of relying on the "release branch" approach?

dietmarw, 2015-08-14:
@bernhard-thiele When you check the download links then you will notice that they they now point to the release page rather than to a direct zip URL. I've made this change some month back when reflecting the release process while working on the new impact paper (it's not a blog but our actual paper ;-) ). So we were thinking alike here. People can still use a release branch to create a modified version but what's probably better is simply create a packaged version and upload that to the release pages including some info. Just like you have done.

tbeu, 2015-08-17:
I adapted the changelog.

bernhard-thiele, 2015-08-17:
@dietmarw That's great, feels more natural than having that release branch. Thanks!

dietmarw, 2015-08-17:
I just noticed that I should have updated the paper for the final version that the new way is already in place ... to late now.

tbeu, 2015-08-17:
Should not be too late if you ask @sjoelund directly.

bernhard-thiele, 2015-08-17:
Yes, just had a chat with @dietmarw and explained, that it can be arranged ...

tbeu, 2015-08-17:
And M_DD is not the only lib considering this ideas. :smirk:

dietmarw, 2015-08-17:
Done: dad21e204c3d96dd24a6ffb472e82dabc9af9459

tbeu, 2015-08-31:
> In case we want to have some information on the new release in the upcoming Modelica newsletter we should meet the deadline September 4th. We can use this issue to discusss the content we want to place there. But time is hurrying...

Any news on this?

bernhard-thiele, 2015-09-01:
There is now a release draft for [1.4.0](https://github.com/modelica/Modelica_DeviceDrivers/releases). 

The zip file was generated from Linux, so line endings are Unix. Should we convert it to windows file endings?

Do you want to make a final check with SimulationX?

tbeu, 2015-09-01:
I do not care about the line endings.

 * Should the copyright be adapted to 2015?
 * Is the help folder location correct or should it rather be in Resources?

bernhard-thiele, 2015-09-01:
Not sure what to do with the copyright notice. I'm not at DLR any more, neither are you. Not sure how this should be handled.

The help folder is the standard folder where Dymola puts generated documentation. From my side that folder is not so important. It was a convention for libraries that were shipped with Dymola to have such a folder. Is there a standardized location for the help folder and should we have such a thing?

tbeu, 2015-09-01:
OK, leave that copyright as it is now.

[Modelica_DeviceDrivers-1.3-build.2.zip](https://github.com/modelica/Modelica_DeviceDrivers/archive/v1.3+build.2.zip) did not contain the help folder. So I was wondering. I do not mind after all.

dietmarw, 2015-09-01:
All non-modelica code is supposed to be places under Resources. So help should be Resources/help. I've told DS so ages ago but doubt they have updated their tool yet. It's only been 6 years or so ;-)

tbeu, 2015-09-01:
But is it needed after all?

dietmarw, 2015-09-01:
`help/`? No. It's just a generated HTML documentation. Some people prefer to have it, though I normally look at the documentation from inside the tool, never with a webbrowser.

bernhard-thiele, 2015-09-01:
I deleted it. Makes the compressed archives a bit smaller.

bernhard-thiele, 2015-09-01:
Can I change the status from Draft to Release?

tbeu, 2015-09-01:
Did you run the tests?

bernhard-thiele, 2015-09-01:
No. Did run some of the tests (I don't have all the required hardware) for the alpha release. Also I currently have no free license to check out. I just hope that things are working as they did before.

tbeu, 2015-09-01:
OK.

Things TODO.
 1. Make the Release then
 1. Edit the Readme.md
 1. Write some short notice for the Modelica newsletter before Sep. 4 and send it to @mtiller.

xogeny, 2015-09-01:
Don't send me the notice.  You need to go to http://newsletter.modelica.org and submit the article through the web.  If you need a login, send an email to webmaster@modelica.org.

tbeu, 2015-09-04:
Can we please get it finished by today? :pray:

bernhard-thiele, 2015-09-04:
Working on it now. Sorry for the last minute actions. I had a high priority date this week, but now that's done and I'm much more relaxed :-). 

I updated the Readme.md with the new release information. Can I state that all non-clocked parts work now with SimulationX?

tbeu, 2015-09-04:
Well with userBufferSize it's all working, but autoBufferSize only works for external solvers CVode and Fixed Step solver and fails for BDF and MEBDF solvers. See https://github.com/modelica/Modelica_DeviceDrivers/issues/54#issuecomment-76032325. I think this might be worth to mention.

tbeu, 2015-09-04:
You mean the newsletter, I guess?

bernhard-thiele, 2015-09-04:
Particularly for the project description (see section "Current release") but I would also mention the improved support of SimulationX in the newsletter.

@tbeu There is now a proposal for the newsletter item in which I added you as "item contributor". That way you should be able to edit that entry, too. Maybe you can have a look whether it looks fine for you and edit it if you find some issues.

tbeu, 2015-09-04:
Saw it, but cannot edit. Can you please edit
 * Many more small fixes, particularly for improved Modelica and tool compatibility.
 * Remove my contribution from the editors. (PR department might feel ignored).

bernhard-thiele, 2015-09-04:
Ok, changed it. 

So we finally have a new release now and I can close that issue :-).

closed 2015-09-04 by bernhard-thiele

tbeu, 2015-09-05:
:white_check_mark::fireworks::beers::one::four::zero::cool: