modelica-3rdparty/Modelica_DeviceDrivers #85: Dimension conflict in SerialPackager

tbeu, 2015-08-05:
I stumbled accross the following lines in the add and get functions of the SerialPackager, e.g.
```mo
    when pkgIn.trigger then
      pkgOut.dummy =
        Modelica_DeviceDrivers.Blocks.Packaging.SerialPackager.Internal.DummyFunctions.addRealAsFloat(
          pkgOut.pkg,
          u,
          pkgIn.dummy,
          byteOrder);
    end when;
```
addRealAsFloat takes a scalar SerialPackager as first argument and returns an scalar Real variable. However pkgOut is not of scalar but of dimension nu.

What is the reason of having nu in PartialSerialPackager?

bernhard-thiele, 2015-08-05:
If I remember correctly, that will allow to connect several sinks at the output, e.g., you could use shared memory and UDP or have UDP sending to two different locations.

However, I just noticed that I can't run some the examples any more. Seems I have brought in some backs with my last renaming action. Need to have a look at this.

tbeu, 2015-08-05:
But still
```mo
partial block PartialSerialPackager
      parameter Integer nu(min=0,max=1) = 0 "Output connector size"
          annotation(Dialog(connectorSizing=true), HideResult=true);
```
looks strange. Either 0 or 1 is allowed but is it not possible to edit in diagram layer.

tbeu, 2015-08-05:
Can't we get rid of nu?

bernhard-thiele, 2015-08-05:
I just made a fix to my previous commit. Didn't rename one variable consistently.

tbeu, 2015-08-05:
OK. But back to my question which is independent of branch.

bernhard-thiele, 2015-08-05:
Regarding the `nu`, this uses the feature of automated connector sizing which should be fine

tbeu, 2015-08-05:
But dimensions are wrong then. It only works because nu is 1 in samples.

bernhard-thiele, 2015-08-05:
The problem is that some of the ports are unconnected at the ends (for the blocks that read). They need to be size 0.

tbeu, 2015-08-05:
OK, the get functions are fine. But problem still is apparent in add functions.

bernhard-thiele, 2015-08-05:
I think I understand it again. The connector is either `1` or `0`. For the reading blocks that is needed so that they can be used without a terminal block. For the writing blocks that is not really needed since they will always be connected to terminal block that writes the package to a (physical) device. However, since the writing blocks inherit the same interface like the reading blocks, they share that code that is not required for them to work (but doesn't hurt either).

bernhard-thiele, 2015-08-05:
An alternative would be to have different base classes for reading or writing blocks (and maybe factor out the common parts in yet another base class). Whether this would be a nicer solution, I'm not so sure.

tbeu, 2015-08-05:
So how did you build the example models? Did you set nu manually in the text layer to 1? This is strange since users are not aware of such hidden parameter.

bernhard-thiele, 2015-08-05:
No, Dymola does set it to 1 automatically as soon as you connect to it. Just tried to build up a new model using drag-and-drop and it worked. Or am I missing some issue?

tbeu, 2015-08-05:
I see. Thus you can easily build a model with nu > 1 and run into trouble.

bernhard-thiele, 2015-08-05:
That's partly true. In theory there is the constraint `max=1` but Dymola doesn't check it. However, in that case the translation will fail due to a wrong number of equations. 

What does SimulationX do? If slight changes would make it more convenient to use that blocks form both SimulationX and Dymola, we could do that.

tbeu, 2015-08-05:
SimulationX 3.6 behaves in the same way as Dymola regarding the automatic incrementation of nu. But it not fail if nu > 1 which is strange.

tbeu, 2015-08-05:
Maybe we schould introduce different connectors for add and get. What do you think?

bernhard-thiele, 2015-08-05:
I will think about it. However, have to stop looking at it and leave know. Of course if you have some idea that we can test, we'll do it...

tbeu, 2015-08-05:
OK. Let's both think about it. Is backward compatibility an serious issue for you?

bernhard-thiele, 2015-08-06:
Backward compatibility is important. There are sometimes good reasons to break it anyway. However, I would prefer starting an alternative packaging system at some day and mark the current one as deprecated.

tbeu, 2015-08-06:
Well, then this issue can be closed, right?

closed 2015-08-06 by bernhard-thiele