modelica-3rdparty/Modelica_DeviceDrivers #86: discrete input/output variables of connector

tbeu, 2015-08-05:
Apply the following patch to the master branch
```diff
 Modelica_DeviceDrivers/Blocks/Interfaces.mo | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/Modelica_DeviceDrivers/Blocks/Interfaces.mo b/Modelica_DeviceDrivers/Blocks/Interfaces.mo
index 76705ef..d667751 100644
--- a/Modelica_DeviceDrivers/Blocks/Interfaces.mo
+++ b/Modelica_DeviceDrivers/Blocks/Interfaces.mo
@@ -5,7 +5,7 @@ package Interfaces
   connector PackageIn "Packager input connector"
     input Modelica_DeviceDrivers.Packaging.SerialPackager pkg;
     input Boolean trigger;
-    input Real dummy;
+    discrete input Real dummy;
 
     output Boolean backwardTrigger;
     output Integer userPkgBitSize;
@@ -27,7 +27,7 @@ package Interfaces
   connector PackageOut "Packager output connector"
     output Modelica_DeviceDrivers.Packaging.SerialPackager pkg;
     output Boolean trigger;
-    output Real dummy;
+    discrete output Real dummy;
 
     input Boolean backwardTrigger;
     input Integer userPkgBitSize;
```
Now Dymola 2016 raises the following translation error on Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_UDP which I do not understand.
```
The following real variables are declared with the prefix discrete,
but are not assigned in a when-clause, either by an assignment or an equation:
addFloat.pkgIn.dummy
addInteger.pkgIn.dummy
addReal.pkgIn.dummy
getFloat.pkgIn.dummy
getInteger.pkgIn.dummy
getReal.pkgIn.dummy
uDPSend.pkgIn.dummy
```

bernhard-thiele, 2015-10-03:
Don't understand it either. Seems to be a wrong error message by Dymola. However, stating that a Real signal is discrete is optional. So it should be no problem to just go without the "discrete" prefix.

tbeu, 2015-10-05:
@HansOlsson I add you here to have a look at the translation error and if the behavior is expected to be right in general. For easy reproduction add in file Modelica_DeviceDrivers/Blocks/Interfaces.mo ```discrete``` to variable ```dummy``` of the two connectors.
Hopefully you can give some feedback. Thank you.

HansOlsson, 2015-10-06:
The error message is correct, and stating that a Real signal is discrete is not optional.

Discrete for a real signal does not say that it is discrete - it states that it is discrete and assigned in a when-loop.

4.4.4 Component Variability Prefixes discrete, parameter, constant
...
**If a Real variable is declared with the prefix discrete it must in a simulation model be assigned in a when-clause, either by an assignment or an equation.** The variable assigned in a when-clause may not be defined in a sub-component of model or block specialized class. [This is to keep the property of balanced models]
A Real variable assigned in a when-clause is a discrete-time variable, even though it was not declared with the prefix discrete. A Real variable not assigned in any when-clause and without any type prefix is a continuous-time variable.
--

tbeu, 2015-10-06:
Thus to make it right for the connectors the connect equations should be in a when-clause?

tbeu, 2015-10-06:
Or can we have the `discrete` only at the `PackageOut` conenctor?

HansOlsson, 2015-10-06:
Cannot put connect-equations in when-clauses:

9.3 Restrictions of Connections and Connectors
The connect-equations (and the special functions for overdetermined connectors) may only be used in equations and may not be used inside if-equations with non-parametric condition, or in when-equations.

I haven't checked if only having discrete at outputs would be correct if assigned in the models.
However,  having 'discrete' at the Interface-level for outputs means that you cannot build hierarchical models using the interface, which seems like a big restriction.

tbeu, 2015-10-06:
Thanks, Hans!

The reason I ask and I applied this `discrete` patch is the following: In new SimulationX 3.7 this (and more) example models give now translation warnings. It is likely that the messages are wrong but I tried to get rid of them which turned out to be not successful.

```
Warning In Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_UDP

The structure of the initial equation system is singular.Equations are missing.
getInteger.pkgIn.dummy:
| getFloat.pkgOut.dummy[1] t--> getFloat.pkgOut.dummy[1]=getInteger.pkgIn.dummy;

There is no unique solution for the variable getInteger.pkgIn.dummy. If applicable, add a fixed initial value, initial equation or initial assignment.

There is no start attribute of state getInteger.pkgIn.dummy. It is set to 0.
```

tbeu, 2015-10-06:
The follwing patches that set pkgIn.dummy.start and pkgIn.dummy.fixed attributes seem to resolve these warnings.

```diff
 Modelica_DeviceDrivers/Blocks/Packaging/SerialPackager.mo | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/Modelica_DeviceDrivers/Blocks/Packaging/SerialPackager.mo b/Modelica_DeviceDrivers/Blocks/Packaging/SerialPackager.mo
index 36a4928..501bd9d 100644
--- a/Modelica_DeviceDrivers/Blocks/Packaging/SerialPackager.mo
+++ b/Modelica_DeviceDrivers/Blocks/Packaging/SerialPackager.mo
@@ -6,7 +6,7 @@ package SerialPackager "Blocks for constructing packages"
     partial block PartialSerialPackager
       parameter Integer nu(min=0,max=1) = 0 "Output connector size"
           annotation(Dialog(connectorSizing=true), HideResult=true);
-      Interfaces.PackageIn pkgIn         annotation (Placement(transformation(
+      Interfaces.PackageIn pkgIn(dummy(start = 0, fixed = true)) annotation (Placement(transformation(
             extent={{-20,-20},{20,20}},
             rotation=180,
             origin={0,108})));
```

```diff
 Modelica_DeviceDrivers/Blocks/Communication.mo | 12 ++++++------
 1 file changed, 6 insertions(+), 6 deletions(-)

diff --git a/Modelica_DeviceDrivers/Blocks/Communication.mo b/Modelica_DeviceDrivers/Blocks/Communication.mo
index 3528517..7aa0631 100644
--- a/Modelica_DeviceDrivers/Blocks/Communication.mo
+++ b/Modelica_DeviceDrivers/Blocks/Communication.mo
@@ -63,7 +63,7 @@ provided by the parameter <b>memoryID</b>. If the shared memory partition does n
       "Buffer size of shared memory partition in bytes (if not deduced automatically)"
       annotation(Dialog(enable=not autoBufferSize, group="Shared memory partition"));
     parameter String memoryID="sharedMemory" "ID of the shared memory buffer" annotation(Dialog(group="Shared memory partition"));
-    Interfaces.PackageIn pkgIn annotation (Placement(
+    Interfaces.PackageIn pkgIn(dummy(start = 0, fixed = true)) annotation (Placement(
           transformation(
           extent={{-20,20},{20,-20}},
           rotation=90,
@@ -157,7 +157,7 @@ provided by the parameter <b>memoryID</b>. If the shared memory partition does n
       annotation (Dialog(group="Outgoing data"));
     parameter Integer port_send=10002 "Target port of the receiving UDP server"
       annotation (Dialog(group="Outgoing data"));
-    Interfaces.PackageIn pkgIn annotation (Placement(transformation(
+    Interfaces.PackageIn pkgIn(dummy(start = 0, fixed = true)) annotation (Placement(transformation(
           extent={{-20,-20},{20,20}},
           rotation=270,
           origin={-108,0})));
@@ -270,7 +270,7 @@ See <a href=\"modelica://Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackag
       "set parity (0 - no parity, 1 - even, 2 - odd)"
        annotation (Dialog(group="Outgoing data"));
 
-    Modelica_DeviceDrivers.Blocks.Interfaces.PackageIn pkgIn annotation (
+    Interfaces.PackageIn pkgIn(dummy(start = 0, fixed = true)) annotation (
         Placement(transformation(
           extent={{-20,-20},{20,20}},
           rotation=270,
@@ -323,7 +323,7 @@ See <a href=\"modelica://Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackag
       "Buffer size of message data in bytes." annotation(Dialog(group="Outgoing data"));
     parameter Integer inputBufferSize=16*1024
       "Buffer size of message data in bytes." annotation(Dialog(group="Incoming data"));
-    Interfaces.PackageIn pkgIn annotation (Placement(transformation(
+    Interfaces.PackageIn pkgIn(dummy(start = 0, fixed = true)) annotation (Placement(transformation(
           extent={{-20,-20},{20,20}},
           rotation=270,
           origin={-108,0})));
@@ -472,7 +472,7 @@ See <a href=\"modelica://Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackag
       parameter Integer ident(min=0) "Identifier of CAN message (CAN Id)";
       parameter Integer dlc(min=0,max=8) = 8
         "Data length code (payload of data in bytes, max=8)";
-      Modelica_DeviceDrivers.Blocks.Interfaces.PackageIn pkgIn
+      Interfaces.PackageIn pkgIn(dummy(start = 0, fixed = true))
         annotation (Placement(transformation(extent={{-20,-128},{20,-88}})));
     protected
       Integer objectNumber;
@@ -703,7 +703,7 @@ See <a href=\"modelica://Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackag
       parameter Integer can_id(min=0) "Identifier of CAN message (CAN Id)";
       parameter Integer can_dlc(min=0,max=8) = 8
         "Data length code (payload of data in bytes, max=8)";
-      Modelica_DeviceDrivers.Blocks.Interfaces.PackageIn pkgIn
+      Interfaces.PackageIn pkgIn(dummy(start = 0, fixed = true))
         annotation (Placement(transformation(extent={{-20,-20},{20,20}},
             rotation=-90,
             origin={-108,0})));
```

tbeu, 2015-10-06:
OK, this is now a diferent issue about the not fully specified initial system.

closed 2015-10-06 by tbeu

HansOlsson, 2015-10-06:
Adding fixed start-values for *inputs* does not make sense to me; I noticed there is a different issue about the initial part and will look at that.