modelica-3rdparty/Modelica_DeviceDrivers #91: Only clocked/sampled RandomRealSource

tbeu, 2015-08-13:
There actually is no need to have sampled RNG in Blocks.OperatingSystem.RandomRealSource. I propose to add a conditional external trigger port for the block. In case this trigger is off the RNG should happen in every step (which matches also the block icon). Do you agree?

tbeu, 2015-08-13:
Same argument would hold for input devices. Of course it would break the backward compatibility.

bernhard-thiele, 2015-08-14:
@tbellmann Would that create any problems with existing code for you?

tbellmann, 2015-08-14:
In most cases, the sample rate is left unchanged, so it wouldn't really that big of an issue. But I don't like external trigger ports in general, I would prefer a solution with a checkbox, like an "disableTrigger" parameter and no additional port. In this case, it also would be completely backward compatible.