modelica-3rdparty/Modelica_DeviceDrivers #92: How to use Arduino Uno

manasdas17, 2015-09-05:
I tried to connect arduino uno board with Openmodelica and Modelica_DeviceDrivers ,but couldn't succeed.Could you lz how to perform the communication.

tbeu, 2015-09-05:
I can't speak for OpenModelica but the serial port communication should work in general to receive and transmit serial data to/from the Arduino Uno.

bernhard-thiele, 2015-09-07:
OpenModelica doesn't  yet support external objects in connection equations. This is needed for the communication blocks. Because of that, OpenModelica's support of the library is currently rather limited. I hope that this will improve at some time, but it will need considerable effort (either to change OpenModelica or to change the packaging system).

closed 2015-09-07 by bernhard-thiele

tbeu, 2015-09-07:
Maybe it would be nice to have some raw functions for the commucation blocks that avoid the packager external object and just deal with vector of doubles or ints.