modelica-3rdparty/Modelica_DeviceDrivers #94: Please test with SimVis

tbeu, 2015-09-24:
@tbellmann: Can you please verify that recent M_DD v1.4.2 works on DLR together with SimVis? Otherwise there might be need to add some defined in same way as https://github.com/DLR-SR/Noise/issues/34#issuecomment-125771660 for the C code of the Noise lib.

@MartinOtter: For reference.

tbeu, 2015-09-24:
Well, there probably is one
```c
#if !defined(WIN32_LEAN_AND_MEAN)
#define WIN32_LEAN_AND_MEAN
#endif
```
missing before
```c
#include <windows.h>
```
in MDDSoftingCAN.h. Can test it next week if it still compiles then.

tbellmann, 2015-09-29:
A short test under windows combining the Version: 1.4.2, 2015-09-14 of the MDD with 1.4 Beta Visualization library worked fine. I tested all UDP and Shared Memory blocks, as well  as most of the input devices. Not testet: CAN and TCP.

commit 6675dc66 (2015-09-30): Fix #94

Define WIN32_LEAN_AND_MEAN before inclusion of Windows.h. Include rpc.h and rpcndr.h afterwards (which are excluded from Windows.h if WIN32_LEAN_AND_MEAN is defined).
closed 2015-09-30 by tbeu