modelica-3rdparty/Modelica_DeviceDrivers #96: Error when running example TestSerialPackager_TCPIP

bernhard-thiele, 2015-10-03:
When running example `Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_TCPIP` in Dymola 2016 FD01 Beta 3 on my Windows 7 virtual machine (32bit) I get following error:

```
The following error was detected at time: 0
MDDTCPIPSocket.h: Unable to connect to server!

The stack of functions is:
Modelica_DeviceDrivers.Communication.TCPIPSocketClient_.connect_
Modelica_DeviceDrivers.Communication.TCPIPSocketClient_.connect_(
tcpipClient.socket, 
"127.0.0.1", 
tcpipClient.port)
Error: Failed to start model.
```
@tbeu I planned to create a new release with your compatibility fixes for SimVis, but would be good if that issue could be resolved, too. Or is it just a problem for my particular set up?

tbeu, 2015-10-04:
Did you also start the server process?

bernhard-thiele, 2015-10-04:
No, just clicked the simulate button...

It works if the server process is started.

closed 2015-10-04 by bernhard-thiele

tbeu, 2015-10-04:
OK.

Please adapt version in package.mo and lib.mos before tagging.