modelica-3rdparty/Modelica_DeviceDrivers #97: TestSerialPackager_UDP: Initial conditions are not fully specified

tbeu, 2015-10-06:
TestSerialPackager_UDP gives translation warnings about the initial conditions being not fully specified. See also #69 where we tried to fix this.

Let me check it and see if I can provide a PR later.

HansOlsson, 2015-10-06:
Using pedantic mode in Dymola the following are reported:
Assuming fixed default start value for the discrete non-states:
addFloat.pkgOut[1].dummy(start = 0.0)
addInteger.pkgOut[1].dummy(start = 0.0)
addReal.pkgOut[1].dummy(start = 0.0)
getFloat.dummy(start = 0.0)
getFloat.pkgOut[1].dummy(start = 0.0)
getInteger.dummy(start = 0.0)
getReal.dummy(start = 0.0)
getReal.pkgOut[1].dummy(start = 0.0)
packager.pkgOut.dummy(start = 0.0)
uDPReceive.pkgOut.dummy(start = 0.0)
uDPSend.dummy(start = 0.0)

To me it seems logical that we need a start-value for dummy if we have:
  when ...trigger then
      dummy=...
  end when;

Using dummy(start=0, fixed=true); would fix that; and the logic is that you should give a start-value for every sampled variable.

tbeu, 2015-10-06:
Thanks, Hans. I'll look at it and check if this also resolves the issue in SimulationX 3.7.

commit c4b93901 (2015-10-06): Fully specify initial conditions for dummy

Fix #97.
closed 2015-10-06 by tbeu