modelica-3rdparty/Modelica_DeviceDrivers #99: Incorrect use of enumeration in Modelica_DeviceDrivers.Blocks.HardwareIO.Comedi.PhysicalData*

HansOlsson, 2015-10-06:
The magic lines
```
converterUnit = cUnit + 1; // convert magic int to readable Modelica enumeration value
```
in Modelica_DeviceDrivers.Blocks.HardwareIO.Comedi.PhysicalDataRead and Modelica_DeviceDrivers.Blocks.HardwareIO.Comedi.PhysicalDataWrite
are not correct Modelica.

It can be corrected by:
```
protected
  final Types.ConverterUnit DummyArray[:]=Types.ConverterUnit.UNIT_volt:Types.ConverterUnit.UNIT_none;
equation
converterUnit = DummyArray[cUnit + 1]; // convert magic int to readable Modelica enumeration value
```

A cleaner alternative is:
```
={i for i in Types.ConverterUnit};
```

(For some odd reason we have 'Integer' to convert from enumeration to integer, but nothing for the reverse.)

HansOlsson, 2015-10-06:
Note that I have not checked the compatibility for the cleaner alternative.

closed 2015-10-07 by tbeu

HansOlsson, 2015-10-07:
To me the choice depends on three factors - and I assume the right choice was made:
* Is it likely that the enumeration will get new members? In that case an array is more robust; if no change is expected it doesn't matter.
* What to do in case the input is not valid? The array-variant will generate an error, the nested-if will generate one specific value. Which one to prefer depend on the application.
* Size of the enumeration. With 3 elements I agree that an array can be too much.

tbeu, 2015-10-07:
OK (for choosing the right implementation choice).

> What to do in case the input is not valid? The array-variant will generate an error.

Yes, that was also my concern. I guess this error can be any kind of a runtime failure leading to unexpected behaviour.

HansOlsson, 2015-10-07:
An array-index out of bounds should normally generate an assertion failure (section 10.1 in Modelica 3.3r1). 

Thus assuming the value can be out-of-bounds (possible since ComediLib is not yet at version 1) the choice is between an assertion and silently converting it to another value, and I generally avoid silently converting things in unexpected ways (unexpected errors are better than undiscovered ones.) Obviously there could be documentation saying that it will not happen or that any new value will be treated as unit-less; in which case the nested-if is perfect.

Another possibility would be to explicitly add an assertion (or just assertion-warning).

commit 1c503303 (2015-10-07): Fix HardwareIO

Fix invalid integer to enumeration type conversion

Close #99.