modelica-3rdparty/Modelica_DeviceDrivers #101: Same issue as #98 with using conditionally enabled variable

HansOlsson, 2015-10-08:
Using conditional variable procPrio is also an issue in
Modelica_DeviceDrivers.ClockedBlocks.OperatingSystem.SynchronizeRealtime
(same solution should apply here).

tbeu, 2015-10-08:
Thanks for testing and pointing to it. I forgot to merge.

tbeu, 2015-10-08:
Hm, should I have this `Boolean initialized(start=false);` both in the SynchronizeRealtime and its (new) sub-block ProcessPriority?

HansOlsson, 2015-10-08:
As I understand it the reason for 'initialized(start=false)' was to ensure that the process-priority was set before the other call in the algorithm. 

Either that is isn't important (it will have the right priority sometime during initialization; might be close enough - in some systems it might be preferable to enable real-time priority after initialization allowing the initialization to take additional time), or it is important (the other call may fail if incorrect priority) - in which case it will take a bit of extra work to handle it. (Using connectors etc.)

tbeu, 2015-10-08:
Would it also work with initial algorithm? This can also be evaluated more than once (as it currently is with `initialized(start=false)`.

commit 603c3852 (2015-10-08): Fix use of conditionally enabled variable procPrio outside of connect

Introduce new block Modelica_DeviceDrivers.ClockedBlocks.OperatingSystem.SynchronizeRealtime.ProcessPriority

Close #101.
closed 2015-10-08 by tbeu