modelica-3rdparty/Modelica_DeviceDrivers #102: CI @ drone.io

tbeu, 2015-10-09:
@dietmarw @bernhard-thiele I added CI at [drone.io](https://drone.io/github.com/modelica/Modelica_DeviceDrivers) for fun in order to run some of the tests automatically.

I got even test_MDDComedi successfully compiling and linking but on amd64 only. Somehow
```sh
sudo apt-get install -qq -y libcomedi-dev
```
works whereas
```sh
sudo apt-get install -qq -y libcomedi-dev:i386
```
always gives `E: Unable to correct problems, you have held broken packages.` no matter what I do.

dietmarw, 2015-10-09:
@tbeu Does this also happen when you start on a fresh build for i386 without building 64bit before?

tbeu, 2015-10-09:
Yes, it happens already at a fresh build.

tbeu, 2015-10-09:
This is now delegated to drone/drone#1230

tbeu, 2015-10-30:
I asked via the drone.io contact formular some 10 days ago but did not get any reply.

closed 2015-10-30 by tbeu

bernhard-thiele, 2015-10-30:
Cool. Quite an amazing service that you have set up :-)