modelica-3rdparty/Modelica_DeviceDrivers #104: License of C source and header files

tbeu, 2015-10-17:
According to Modelica ticket 1525 the license of the C source and header files should be changed to a more appropriate one.

bernhard-thiele, 2015-10-19:
I just read through the discussion of the quoted ticket. I wouldn't mind to have BSD 2 or BSD 3 for the C sources. We probably would need to ask contributors for that change, though. However, in a nutshell, what is the problem of just using the Modelica license for the C code?

tbeu, 2015-10-19:
See DLR-SR/Noise#9.

tbeu, 2015-10-19:
See also #35.

bernhard-thiele, 2015-10-19:
@tbeu I'm not into the details, but looking at ticket 1525 it seems there is a bias for BSD2 over BSD3, but it was voted for BSD3 at some design meeting. So what would you prefer?

@tbellmann Would it be fine for you to change the license of the external C code to BSD2 or BSD3? Preferences?

tbeu, 2015-10-19:
Yes, there recently was some agreement to get rid of the endorsment clause of BSD3 and therefore switch to BSD2.

tbellmann, 2015-10-22:
Hi,
As long as anyone can use the library freely and without any warrenty regarding DLR or the authors, I am fine with any license. BSD3 sounds Ok to me, more fame for the authors ;)
Greetings from Oberpfaffenhofen,
Tobias

bernhard-thiele, 2015-10-22:
@tbellmann Good, though the fame might be less, I understand that BSD2 would be fine for you, too.

I just looked at the endorsement clause of BSD3 and I wonder what people would need to write if advertising that their product is running with MDD. Maybe they would need to state s.th. like "Modelica_DeviceDrivers (external C code developed by Modelica Association)", but that is not completely obvious, is it?

The BSD license is only one part of the library, while the Modelica part would be under the Modelica license. So BSD2 seems to be simpler to handle. 

I wonder if one could ship the external C with both licenses and allow people to use whatever suits them best, e.g., Modelica License 2 or any later version and BSD2 License. In that way one could drop BSD2 if a future version of the Modelica License doesn't have the issues for external C code that are currently criticised. What do you think?

tbeu, 2015-10-22:
I would not call it "any later version or BSD2 License". In MSL we have MoLic2 for almost all files and explicitly list the exceptions. This should also be approriate here.

tbeu, 2016-01-12:
Please note that Modelica ticket 1525 is now solved.

bernhard-thiele, 2016-03-07:
In order to be close to the MSL I would stick to the Modelica license for the Modelica code and move to MoLic2 for the C code.

tbeu, 2016-03-07:
MSL 3.2.2 changed all C code to Simplified BSD License.

bernhard-thiele, 2016-03-07:
Okay, then let's change to Simplified BSD License

bernhard-thiele, 2016-03-08:
@tbeu @dietmarw  How to do the external C-code license change to _"Simplified BSD License"_ properly?

I just checked the MSL svn, directory `Modelica/Resources/C-Sources` and I didn't find pointers to the Simplified BSD License license. Where is the license stated for MSL external C-code?

What I would do for MDD is to replace

```
@copyright Modelica License 2
```

by s.th. like

```
@copyright Simplified BSD License
```

Does this make sense, or is _"Simplified BSD License"_ not really a set term for a license, but rather a loose classification?

Furthermore, we need to adapt file `Modelica_DeviceDrivers/Resources/License.txt`, I guess we would do this by copying and adapting the text from Wikipedia (https://en.wikipedia.org/wiki/BSD_licenses#2-clause_license_.28.22Simplified_BSD_License.22_or_.22FreeBSD_License.22.29)?

```
Copyright (c) <YEAR>, <OWNER>
All rights reserved.

Redistribution and use in source and binary forms, with or without
modification, are permitted provided that the following conditions are met:

1. Redistributions of source code must retain the above copyright notice, this
   list of conditions and the following disclaimer.
2. Redistributions in binary form must reproduce the above copyright notice,
   this list of conditions and the following disclaimer in the documentation
   and/or other materials provided with the distribution.

THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND
ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT OWNER OR CONTRIBUTORS BE LIABLE FOR
ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
(INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND
ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
(INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

The views and conclusions contained in the software and documentation are those
of the authors and should not be interpreted as representing official policies,
either expressed or implied, of the FreeBSD Project.
```

Does this fit your idea? How was this license text adapted for MSL? Where can I find it?

tbeu, 2016-03-08:
Yes, it is enough to adapt the first copyright line of this license text by addinf year (span) and owners. Then it should be added to all C/H files.

bernhard-thiele, 2016-03-08:
Hm, by this approach I need to make a decision about copyright holders on a per file base. I guess this implies to decide about the threshold of changes within a file that justify adding a copyright holder. I will have a first go based on my memory rather than tracking back repository histories. Feel free to discuss/modifiy my decisions.

tbeu, 2016-03-08:
Well, I thought we agreed on the general copyright year span and ownership. No need for me to have check the commit history of each file.

bernhard-thiele, 2016-03-09:
@tbeu @tbellmann  I have updated the license in 70a0374

The way I have solved that for now is to have one general license file "License.txt" within the Resources folder and references from other files to this general license.

In License.txt I have for the Moment kept Modelica License 2 as one option besides the Simplified BSD. Hence, the C-code is dual licensed. This way one can simply state the Library is Modelica License 2. However, if that license is problematic for some use-cases one can sidestep to the more liberal one.

Comments? Can I close this ticket?

tbeu, 2016-03-09:
Will check it later.

dietmarw, 2016-03-09:
> "if that license is problematic for some use-cases one can sidestep to the more liberal one."

You mean like Modelica License 2 make it actually impossible to fork this library on GitHub and provide PRs?

tbeu, 2016-03-10:
I noticed that you changed the test C files to BSD-2-Clause and the H files under dual license. I propose to change the license of all C/H files to BSD-2-clause.

The MoLic2 might be problematic for the Modelica parts but this is not part of this ticket.

dietmarw, 2016-03-10:
Probably worth mentioning that if you use BSD-2-clause that this is fully compatible with MoLic2. But not the other way round. So not confusing people with dual licensing and simply picking BSD-2-clause simplifies this.

tbeu, 2016-03-10:
Do you mean BSD2-clause for all the library (mo and c code)? I am in favour.

dietmarw, 2016-03-10:
That would be ideal. But please no double-license. People are struggling with licenses as it is.

bernhard-thiele, 2016-03-10:
Ok, I see that dual licensing is not a popular pick. I found it quite elegant since it simply allows people to take what suits them best. It also allows to simply state the the library is MoLic2 which I find less confusing than explaining that the Modelica code is MoLic2, but the C code is BSD-2-clause like...

Particularly since (as I understand now) BSD2 is more an umbrella term for certain clauses within a license rather than a concrete license which was confusing to me.

Actually, I wonder that nobody argued for a dual licensing option in https://trac.modelica.org/Modelica/ticket/1525.

The MSL Github page "avoids" that confusion by stating that the MSL is MoLic2 although now more and more parts within it are actually not MoLic2. In my opinion, this is bad. Dual licensing allows "truthfully" stating that the library is MoLic2 while working around some issues of MoLic2 if needed (hoping that MoLic2 is revised at some point in time).

Changing the C-code to BSD-2 only will make MDD a dual-licensed library anyway with all the confusion that this adds. Allowing both MoLic2 and BSD-2 for the C-code at least allows to more truthfully state that the library is MoLic2 at the project top level.

This is why I find dual licensing something that is worth considering (maybe also for the MSL). I'm fine removing the dual licensing in the C-code if this opinion is not shared, but I would be interested in your arguments.

dietmarw, 2016-03-10:
On 10 March 2016 at 10:59, Bernhard Thiele <notifications@github.com> wrote:
> Ok, I see that dual licensing is not a popular pick. I found it quite elegant since it simply allows people to take what suits them best. It also allows to simply state the the library is MoLic2 [...]

Again I can not stress too much that MoLic2 is in its current state
not suitable to anything but the MSL and even there it causes issues.
BSD doesn't have this problem but basically giving the same freedom of
use.

> Particularly since (as I understand now) BSD2 is more an umbrella term for certain clauses within a license rather than a concrete license which was confusing to me.

BSD2 is an explicit version of the BSD variants. So it is very
accurate in what it means.

> Actually, I wonder that nobody argued for a dual licensing option in https://trac.modelica.org/Modelica/ticket/1525.

I guess using MoLic2 makes no sense for use of non-Modelica files.

> The MSL Github page "avoids" that confusion by stating that the MSL is MoLic2 although now more and more parts within it are actually not MoLic2. In my opinion, this is bad. Dual licensing allows [...]

Don't hold your breath. MO does not really see an issue so...
The GitHub page needs simply an update stating the current license status.

> Changing the C-code to BSD-2 only will make MDD a dual-licensed library anyway with all the confusion that this adds. Allowing both MoLic2 and BSD-2 for the C-code at least allows to more truthfully state that the library is MoLic2 at the project top level.

Well you could simply put it out under BSD2 license only. No need to
dual licensing.

bernhard-thiele, 2016-03-10:
Fixed in 2665d97550901d284311b65d5ba72ac1291159aa

closed 2016-03-10 by bernhard-thiele

bernhard-thiele, 2016-03-10:
Okay, changed C-code to BSD-2 only. Probably some other documentation adjustments are necessary at the top-library level. I will stick for the time being with MoLic2 for the Modelica code.