modelica-3rdparty/Modelica_DeviceDrivers #109: Update Copyright

tbeu, 2016-03-07:
Old Copyright still says 2013 and DLR. This needs to be updated to inlcude LIU and ITI since 2014., i.e.
```html
Copyright &copy; 2012-2013, DLR Institute of System Dynamics and Control
Copyright &copy; 2014-2016, Link&ouml;ping University and ITI GmbH
```
It also needs to be considered for the C/H files when updating the license header in #104.

@tbellmann Is this alright for DLR?

bernhard-thiele, 2016-03-08:
Or maybe in the following way:
```html
Copyright &copy; 2014-2016,  Link&ouml;ping University, ITI GmbH, and DLR Institute of System Dynamics and Control
Copyright &copy; 2012-2014, DLR Institute of System Dynamics and Control
```

tbeu, 2016-03-08:
Really `2012-2014`? Or `2012-2013`?

tbeu, 2016-03-08:
I guess it's fine as you proposed.

But then I need to change https://github.com/svn2github/Modelica_StandardLibrary/blob/master/Modelica/Resources/C-Sources/ModelicaMatIO.c#L3-L4 from
```
   Copyright (C) 2005-2016, Christopher C. Hulbert
   Copyright (C) 2013-2016, Modelica Association and ITI GmbH
```
to
```
   Copyright (C) 2013-2016, Christopher C. Hulbert, Modelica Association and ITI GmbH
   Copyright (C) 2005-2013, Christopher C. Hulbert
```

bernhard-thiele, 2016-03-08:
No, I think in your cases above, both forms are equal. But if you would just write
```
   Copyright (C) 2013-2016, Modelica Association, and ITI GmbH
   Copyright (C) 2005-2013, Christopher C. Hulbert
```
It would appear to me as if Christopher would have transferred his copyrights to Modelica Association and ITI. However, I'm not an expert in this and maybe this interpretation is wrong.

bernhard-thiele, 2016-03-19:
I think I have done all necessary copyright updates. If there are any things that I missed, please let me know.

closed 2016-03-19 by bernhard-thiele