modelica-3rdparty/Modelica_DeviceDrivers #110: File test_TCPIPSocketServerEcho.cpp should go into thirdParty folder

bernhard-thiele, 2016-03-09:
The file 

Modelica_DeviceDrivers/Modelica_DeviceDrivers/Resources/test/Communication/test_TCPIPSocketServerEcho.cpp 

states that is from https://msdn.microsoft.com/en-us/library/windows/desktop/ms737593.aspx

Hence, it should be below thirdParty, shouldn't it?

tbeu, 2016-03-09:
Do not think so. Well, 
- it has undefined license
- and there is no thirdParty folder in the test folder.

bernhard-thiele, 2016-03-09:
Undefined license doesn't mean that it is in the public domain. We can have another directory under thirdParty for it. Legally, that is the cleaner solution. The build/test scripts can be adapted to it. Actually, if it has no license and we don't know if it is considered to be in the public domain we need to think whether we remove it entirely. 

This licensing aspect is very important for tool vendors that ship this library. We cannot take  that lightly.

tbeu, 2016-03-09:
http://stackoverflow.com/questions/2181578/licensing-for-msdn-code-samples

bernhard-thiele, 2016-03-09:
Okay, when we have " Microsoft Limited Public License" for this code. As I understand it this makes it fine to include it under thirdParty together with a copy of the license since it states

```
If you distribute any portion of the software in source code form, you may do so only under this license by including a complete copy of this license with your distribution
```

tbeu, 2016-03-09:
New folder thirdParty in folder test or existing folder thirdParty?

bernhard-thiele, 2016-03-10:
The existing folder thirdParty. The idea is to collect all third party code there, so that it is separated from the rest and particularly vendors have an easy overview about it when checking licenses on whether they can ship it safely.

tbeu, 2016-03-10:
OK, then put it in thirdParty\Microsoft

bernhard-thiele, 2016-03-10:
Good, that fits well

bernhard-thiele, 2016-03-19:
Missed that this was is not yet resolved and just did the release. This should be resolved, but probably not so severe that we immediately need a 1.4.5 release

commit fedfb709 (2019-09-17): Moved TCPIPSocketServerEcho.cpp to thirdParty

Closes #110.
closed 2019-09-17 by bernhard-thiele