modelica-3rdparty/Modelica_DeviceDrivers #114: Compilation error in serial port communication.

sswaroopk, 2016-04-30:
Hi,
I'm trying to communicate with serial port using Modelica_DeviceDrivers package on windows. I have given all the inputs required for the functions in it. But when I simulate it, I get many compilation errors with "undefined reference to `_imp__pthread" .Please help me on this.

-Swaroop.

tbeu, 2016-04-30:
Can you attach your sample code here pls.

tbeu, 2016-04-30:
You may also have a look at test_MDDSerialPort.c.

sswaroopk, 2016-04-30:
This is the sample code
model Try2
  parameter String Device_name = "COM4";
  import Modelica_DeviceDrivers.Communication.SerialPort;
  import Modelica_DeviceDrivers.Utilities.Types.SerialBaudRate;
  import Modelica_DeviceDrivers.Utilities.Types.ByteOrder;
  import Modelica_DeviceDrivers.Communication.SerialPort_;
  import Modelica_DeviceDrivers.Packaging.SerialPackager;
  import Modelica_DeviceDrivers.Packaging.SerialPackager_;
  parameter SerialBaudRate baud = SerialBaudRate.B9600;
  parameter ByteOrder byteorder = ByteOrder.LE;
  Modelica_DeviceDrivers.Communication.SerialPort OpenPort;
  Modelica_DeviceDrivers.Communication.SerialPort ClosePort;
  Modelica_DeviceDrivers.Packaging.SerialPackager pkg;
equation
  OpenPort = SerialPort.constructor(Device_name, 16 * 1024, 0, 1, baud);
  pkg = SerialPackager.constructor();
  SerialPort_.read(OpenPort, pkg);
end Try2;
This is the error. I have also tried opening a port using blocks. But getting the same error.
[image]

tbeu, 2016-04-30:
I fixed the model and renamed to Try3:
```mo
model Try3
  parameter String Device_name = "COM4";
  import Modelica_DeviceDrivers.Communication.SerialPort;
  import Modelica_DeviceDrivers.Utilities.Types.SerialBaudRate;
  import Modelica_DeviceDrivers.Utilities.Types.ByteOrder;
  import Modelica_DeviceDrivers.Communication.SerialPort_;
  import Modelica_DeviceDrivers.Packaging.SerialPackager;
  parameter SerialBaudRate baud = SerialBaudRate.B9600;
  parameter ByteOrder byteorder = ByteOrder.LE;
  SerialPort OpenPort = SerialPort(Device_name, 16 * 1024, 0, 1, baud);
  SerialPackager pkg = SerialPackager();
equation
  SerialPort_.read(OpenPort, pkg);
  annotation(uses(Modelica_DeviceDrivers(version="1.4.4")));
end Try3;
```
This model works with C Code-Export of SimulationX 3.7 and compilation with MinGW, Cygwin or MSVC. The pthread functions are not used by this model if compiled on Win platform with MinGW, Cygwin or MSVC. You can check if you add in MDDSerialPort.h after line 21 a test like `#error Test MinGW`. If you see this message in the compiler log you can be sure that the correct implementation of the serial port communication (i.e., no pthreads dependency) is taken. And therefore I assume it might be related to OM 1.9.3 rather then to the library.

tbeu, 2016-04-30:
You might also try to remove pthread.dll from Modelica_DeviceDrivers\Resources\Library\win32 and see if something changes.

sswaroopk, 2016-05-02:
Hi,
Thanks for the reply. Removing pthread.dll did it.
Now I'm able to open the port and send data successfully. However if I wanted to receive from the port nothing comes in. Below are the sample codes for sending and receiving data respectively.
`model arduinosend
  parameter String Device_name = "COM5";
  import Modelica_DeviceDrivers.Communication.SerialPort;
  import Modelica_DeviceDrivers.Utilities.Types.SerialBaudRate;
  import Modelica_DeviceDrivers.Utilities.Types.ByteOrder;
  import Modelica_DeviceDrivers.Communication.SerialPort_;
  import Modelica_DeviceDrivers.Packaging.SerialPackager;
  import Modelica_DeviceDrivers.Packaging.SerialPackager_;
  parameter SerialBaudRate baud = SerialBaudRate.B9600;
  parameter ByteOrder byteorder = ByteOrder.LE;
  SerialPort OpenPort = SerialPort.constructor(Device_name, 16 * 1024, 0, 0, baud);
  SerialPackager pkg = SerialPackager.constructor(4608);
  parameter Integer S1[1] = {50};
  algorithm
  SerialPackager_.setPos(pkg, 0);
  SerialPackager_.addInteger(pkg, S1, byteorder);
  SerialPort_.sendTo(OpenPort, pkg, 4608);  
end arduinosend;`

model for receiving
`model arduinoread
  parameter String Device_name = "COM5";
  import Modelica_DeviceDrivers.Communication.SerialPort;
  import Modelica_DeviceDrivers.Utilities.Types.SerialBaudRate;
  import Modelica_DeviceDrivers.Utilities.Types.ByteOrder;
  import Modelica_DeviceDrivers.Communication.SerialPort_;
  import Modelica_DeviceDrivers.Packaging.SerialPackager;
  import Modelica_DeviceDrivers.Packaging.SerialPackager_;
  parameter SerialBaudRate baud = SerialBaudRate.B9600;
  parameter ByteOrder byteorder = ByteOrder.LE;
  SerialPort readPort = SerialPort.constructor(Device_name, 16 * 1024, 0, 1, baud);
  SerialPackager pkg = SerialPackager.constructor(4608);
  //SerialPackager pkg1 = SerialPackager.constructor(4608);
  //parameter Integer S1[1] = {50};
  //Real n;
  Integer I[16];
  algorithm
  SerialPackager_.setPos(pkg, 0);
  SerialPort_.read(readPort,pkg);
  //n=SerialPort_.getReceivedBytes(readPort);
  I:=SerialPackager_.getInteger(pkg,16,byteorder);
end arduinoread;`

Also does this package support bidirectional communication (I'm getting 'process crashed' error if I try to send and receive in the same model)?

tbeu, 2016-05-02:
Well, Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_SerialPort uses send/receive in the same model. Last time I tested it, it worked.

tbeu, 2016-05-03:
Just tested TestSerialPackager_SerialPort with com0com emulator on Win again. Works as expected.

I cannot figure out what your problem with model arduinoread is. Sorry.

@bernhard-thiele Can you please reproduce the strange linking error against MSVC win32/phthread.dll with 32-bit MinGW/OpenModelica on Windows. I once created an official OpenModelica ticket that I consider it rather strange that OM tries to link against MSVC libs. But this time it is pthread.dll and not pthread.lib which causes the original problem reported by @sswaroopk.

sswaroopk, 2016-05-03:
Thanks for the reply. Seems like I'm doing something wrong somewhere then.
Actually the above sample code I wrote for an arduino which, when opened gives 100 values with some delay serially. But when I try running the code I'm getting nothing.
Also the 'process crashed' error comes when I try to send and receive with same port (creating two instances, for reading and sending)
And whenever I use RealTimeSynchroniation, it crashes and exits with code-1073741701 (Not sure whether this is related to win 10)
Sorry for vexing.

tbeu, 2016-05-03:
Can you please modify the test model test_MDDSerialPort.c in such a way that the Arduino in one setup receives the values from Windows and in the the other setup send the values. This is the minimum test you can do to see if you get serial port communication working. Thanks.

bernhard-thiele, 2016-05-03:
@tbeu I just looked into it a bit and I don't know a good way to fix the linking problem. Standard compilation with OM relies on `-lpthread` and OM ships with its own flavour(s) of `pthread.dll`. If one provides a dummy mingw library for Windows the mingw linker will use the dummy library instead of the pthread system library. Consequently, it complains about missing symbols.

As a workaround I created a folder `Library\mingw32` and copied one of the pthread libraries that ship with OM (`pthreadGC2.dll`) into the directory and renamed it to `pthread.dll`. This made it build, but it is, of course, not a nice solution. I'm not sure whether to commit that. Maybe there is a better solution.

tbeu, 2016-05-03:
@bernhard-thiele What I do not understand is why OM uses pthread.dll during linking by -lpthread. pthread.dll is a (dummy) dynamic link library. What I would expect is that the MinGW linker uses a libpthread.a. Can you please investigate this? Is there any libpthread.a from OM? Does it help if you put this libpthread.a in Resources/Library/win32 or ming32?

@sswaroopk If a serial port is to be send and receive data it has to be constructed by
```mo
SerialPort readWritePort = SerialPort(Device_name, 16 * 1024, 0,
  1 /* must be non-zero for bidirectional communication */,
  baud);
```

tbeu, 2016-05-04:
@bernhard-thiele @sjoelund I just noticed that ModelicaStandardTables, ModelicaRandom and ModelicaInternal of the MSL also inlude pthread.h for POSIX systems and thus depend on pthread library. However, the corresponding Modelica blocks do not declare an explicit library dependency to pthread library. If this is correct, then for M_DD the solution is to remove all explicit dependencies on pthread from the Library annotations and to remove the dummy binary pthread.dll. What do you think?

bernhard-thiele, 2016-05-04:
Just checked with Dymola on Linux. There is a linker error if the pthread library dependency is not declared in the library annotations.

I have no experience with MinGW. However, I don't find it surprising that it uses pthread.dll if `-lpthread` is one of its argument. The gcc toolchain doesn't need a static import library for using shared libraries on linux. MinGW is a port of gcc and GNU Binutils to windows and apparently can do the same trick for DLLs (understands DLLs directly, http://stackoverflow.com/questions/3573475/how-does-the-import-library-work-details).

We could use the workaround of providing a copy of the pthread library for MinGW in the `Library\mingw32` folder. The disadvantage is that the pthread libraries shipped with omc might vary with different omc versions which could give rise to some surprising behaviour.

@sjoelund how sane is it to put a pthread library into the library resources?

sjoelund, 2016-05-04:
Not very. It could conflict with the system libraries. What is using pthreads in the library? One of the dependencies?

tbeu, 2016-05-04:
> Just checked with Dymola on Linux. There is a linker error if the pthread library dependency is not declared in the library annotations.

Does this also mean that the MSL Tables, Utilities or Noise functions also give a linker error in Dymola on Linux since they also require the pthread library?

tbeu, 2016-05-04:
> What is using pthreads in the library?

The UDP, the serial port and the socket CAN implementation require pthread (on Linux only). On Windows (no matter if MinGW or MSVC) pthread is not required. However, for a platform-generic implementation the pthread library is added as dependency in the external Library annotation and a dummy pthread.dll (without any symbols) is provided - for MSVC only.

bernhard-thiele, 2016-05-04:
> Does this mean that the MSL Tables, Utilities or Noise functions also give a linker error in Dymola on Linux since they also require the pthread library?

No, it MSL Tables or Noise works fine. As experiment I also used tables and the UDP blocks in the same model. This still gives a linker error if the UDP external functions don't declare the pthread dependency:

````
/usr/bin/ld: /tmp/ccIDqvLV.o: undefined reference to symbol 'pthread_create@@GLIBC_2.1'
//lib/i386-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
```

MSL Tables or Noise only use mutex related functions (no thread creation), maybe these bindings are provided somehow, but the thread creation bindings are not provided. I don't know.

sjoelund, 2016-05-04:
What is the full gcc command invoking the linker? Perhaps they just add -lpthread at a static position in the command-line so that it is only visible to the "standard" libraries?

bernhard-thiele, 2016-05-04:
Ok, translating following model

```modelica
model Unnamed
  Modelica.Blocks.Tables.CombiTable1Ds combiTable1Ds(table=[0.0,0.0; 1,1]);
  Modelica.Blocks.Sources.Clock clock;
equation 
  connect(clock.y, combiTable1Ds.u);
end Unnamed;
```

and tweaking the output of the Dymola build script to print the gcc command, I get following output: 

```
cc -O0 -Wl,--no-as-needed -o dymosim  -DDYMOSIM -DUNIX -DSkip_f2c_Undefs -m32   -Wl,--no-undefined -Wl,--allow-multiple-definition -I.  -I/opt/dymola-2017-x86_64/source dsmodel.c /opt/dymola-2017-x86_64/source/matrixop.c  -L'/opt/dymola-2017-x86_64/Modelica/Library/Modelica 3.2.2/Resources/Library/linux32' -L/opt/dymola-2017-x86_64/bin/lib  -lModelicaMatIO -lModelicaStandardTables -lzlib  -lModelicaMatIO -lModelicaStandardTables -lzlib -lds -lstdc++ -lm -ldl
```

sjoelund, 2016-05-04:
Are the MSL libraries dynamically linked? (`ldd dymosim`)

bernhard-thiele, 2016-05-04:
`ldd dymosim`:

```
        linux-gate.so.1 =>  (0xf7718000)
        libds.so => not found
        libstdc++.so.6 => /usr/lib/i386-linux-gnu/libstdc++.so.6 (0xf75fc000)
        libm.so.6 => /lib/i386-linux-gnu/libm.so.6 (0xf75b5000)
        libdl.so.2 => /lib/i386-linux-gnu/libdl.so.2 (0xf75b0000)
        libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xf7402000)
        /lib/ld-linux.so.2 (0xf7719000)
        libgcc_s.so.1 => /lib/i386-linux-gnu/libgcc_s.so.1 (0xf73e5000)
```

tbeu, 2016-05-04:
One more idea: Keep the Library annotation as it is but remove the pthread.dll from the M_DD distribution. Of course this needs to be checked with Dymola on Windows (MSVC and MinGW). For SimulationX (MSVC) it is not an issue. It looks for pthread.lib and if it is not found the linker dependency simply is ignored.

bernhard-thiele, 2016-05-04:
If Dymola doesn't find a library dependency it is an error (at least for MSVC, haven't checked for MinGW). In order to make the linker happy I came up with the "dummy" libraries. Tricky to find a solution that works for several tools and platforms ...

tbeu, 2016-05-05:
@bernhard-thiele Thanks for checking with Dymola.

> Would it help if the pthread.dll is not a dummy lib but exports the required symbols just for the linker?

I guess this could be the solution then. The dummy libs should export the required symbols - w/o a proper implementation though.

tbeu, 2016-05-06:
@bernhard-thiele Are you going to set up a pthread.dll with the exported symbols required for MinGW linking?

tbeu, 2016-07-14:
This was a hot topic back in early days of May but is not yet finalized. @bernhard-thiele

Again: My idea is instead of having a dummy pthread.dll w/o any exported symbols a modified dummy pthread.dll with exactly the symbols (pthread_mutex_init, pthread_mutex_lock etc.). This pthread.dll shall still be located in the win32/win64 directories and guids the MinGW linker only. Does it make sense? Shall I try it?

bernhard-thiele, 2016-07-14:
Yes, please try it. I'm not working with the MinGW tool chain so there is not much experience for it from side. However, I would expect problems. if you export the symbols, wouldn't any binary that links to it try to use these symbol addresses as targets for jumping into the respective function? But maybe, I'm wrong and after you've tried it we will know...

tbeu, 2016-07-14:
AFAIU the pthread.dll is only used for the MinGW linker and not at runtime.

bernhard-thiele, 2016-07-14:
Regarding MDD, yes. But the libraries are linked with the omc simulation executable, and if the omc runtime uses the same pthread functions, it might jump into the dummy library instead of jumping into the correct library. But, I might be wrong with this...

tbeu, 2016-07-14:
I thought that the MinGW linker (triggered from MC) uses the wrong dir but at runtime the correct MinGW system lib is used.

Far better idea: For another project we used a real pthread.dll compiled from https://github.com/GerHobbelt/pthread-win32 with VS. It was only 59kB for the x64 version (32-bit version not yet built). I propose that I build pthread.dll (manually at my PC with VS2010) and commit the proper files. What do you think?

bernhard-thiele, 2016-07-14:
I think I discussed an idea like that with @sjoelund somewhere in the middle of this thread and his response to asking whether this idea is sane was

> Not very. It could conflict with the system libraries. What is using pthreads in the library? One of the dependencies?

tbeu, 2016-07-14:
@sswaroopk Can you please test if c8b9dd6f4a29e84f365700aa05bcdce415af0abe works for you. Thanks.

tbeu, 2016-07-14:
I am not that optimistic if @sswaroopk is active and able to reproduce. Any chance to reproduce at LIU?

bernhard-thiele, 2016-07-14:
I will be more or less offline for the next days and I'm currently also a bit reluctant to invest too much time diving into the Windows MDD MinGW world. But of course it would be great if you managed to find a solution that we can integrate into MDD.

tbeu, 2016-07-14:
Thanks for letting me know.

Wood be good to the/some communication examples woking with OM on Win. I asked @dietmarw if he can reproduce and test.

tbeu, 2016-07-15:
@dietmarw Another test case
```mo
model Try4
  parameter String mem_name = "COM4";
  import Modelica_DeviceDrivers.Communication.SharedMemory;
  import Modelica_DeviceDrivers.Communication.SharedMemory_;
  import Modelica_DeviceDrivers.Packaging.SerialPackager;
  import Modelica_DeviceDrivers.Packaging.SerialPackager_;
protected
  SharedMemory sm = SharedMemory(mem_name, 16 * 1024);
  SerialPackager pkg = SerialPackager();
equation
  SharedMemory_.read(sm, pkg);
  annotation(uses(Modelica_DeviceDrivers(version="1.4.4")));
end Try4;
```

tbeu, 2016-07-15:
@sjoelund Is there any chance to tell the linker to first search for pthread.dll in the MinGW system dirs and afterwards in the Library/win32 dir? WouldN#t this be the most general solution?

tbeu, 2016-07-15:
@sjoelund dietmarw could reproduce the MinGW linker error and confirm that linking works after c8b9dd6 was applied. However, running the model Try3.exe or Try4.exe was not successful due to application crash. We could not figure out if this is due to the pthread.dll or not.
```
stdout | OMEditInfo | <p>C:/Users/dietmarw/AppData/Local/Temp/OpenModelica/OMEdit/Try3.exe -port=65159 -logFormat=xml -override=startTime=0,stopTime=1,stepSize=0.002,tolerance=1e-6,solver=dassl,outputFormat=mat,variableFilter=.* -r=Try3_res.mat -dasslJacobian=coloredNumerical -w -lv=LOG_STATS</p>
stdout | error | <p>Process crashed<br>
Simulation process failed. Exited with code -1073741515.</p>
```

tbeu, 2016-08-05:
Since @sswaroopk no longer is busy, can someone from the OM developers on Windows try to debug/reproduce the crash?

tbeu, 2016-10-16:
@vwaurich It looks like OpenModelica/OMCompiler#1102 exactly addresses this issue. Can you test on Windows if Try3.exe and Try4.exe (from my models given above) work with current M_DD master (!) in you OM env. Thanks a lot!

vwaurich, 2016-10-17:
Try3 compiles, since omc does not refer to the ressource directory of external pthread versions. Try4 does not compile since it uses rt library as well and therefore the ressource directory is provided to the linker which finds the pthread.dll in this directory and still prefers the MDD-pthread instead of the omc-pthread.

tbeu, 2016-10-17:
@vwaurich Thanks for testing. Does Try3.exe also run without application crash? I thought you exclude the MDD-pthreads.dll from linking if the resource directory is provided to the linker.

vwaurich, 2016-10-17:
Tested Try3 with an arduino. It creates the serial connection without crashing and finishes simulation. The MDD resources directory is excluded but another pthread.dll is linked by omc by default.

tbeu, 2016-10-19:
@vwaurich I hopefully found a solution by daef39aaf6ddb7e2e82a350eb411b41ab5e02738. Can you please switch to git branch pthread-win32 and check again Try3 and Try4 with OMC on Win using MinGW. Thanks a lot.

vwaurich, 2016-10-20:
@tbeu Doesnt work. Still get the '<undefined reference to pthread_mutex_trylock>'. But I can restrict OMC to not link rt under windows. This fixes the Try4 and linking of SharedMemory. Although, <'Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_SharedMemory'> also compiles and starts to simulate then, it crashes with some runtime error. Maybe thats another OMC bug.

vwaurich, 2016-10-20:
Best thing would be to force the linker to ignore certain libs that can be found among all specified library dirs. But I dont know how. The order of given of -L and -l commands makes no difference.

tbeu, 2016-10-20:
Why does it look for pthread_mutex_trylock (and other)? It is not needed at all on Win.

vwaurich, 2016-10-20:
Its not needed for MDD but OMC runtime needs it.

tbeu, 2016-10-20:
I tried yesterday to neither have a pthread.lib nor pthread.dll in Library\win32 but then in Dymola the MSVC linker complains about the missing explicit dependency.

Can you please try to put your OM libpthread.a/pthread.lib to Library\mingw32? Does linking work? See Bernhards earlier try where he tried it with the dynamic link lib instead.

> As a workaround I created a folder Library\mingw32 and copied one of the pthread libraries that ship with OM (pthreadGC2.dll) into the directory and renamed it to pthread.dll. This made it build, [...]

vwaurich, 2016-10-20:
If I replace pthread.lib with libpthreadGC2.a from the OMDev - repo or pthread.dll with pthreadGC2.dll as bernhard did, I end up with a new undefined reference: `<pthread_getclean>` .

tbeu, 2016-10-20:
@bernhard-thiele Can you reproduce @vwaurich's observation?

bernhard-thiele, 2016-10-20:
I have no ready-configured Windows machine for testing, but I will install stuff and give it a try over the weekend.

bernhard-thiele, 2016-10-22:
I also get the undefined reference: `<pthread_getclean>` error.

tbeu, 2016-10-24:
@sjoelund Would it be possible to change the library search order by first searching in the OM run-time path, then Library\mingw32 and last Library\win32?

tbeu, 2016-10-24:
@sjoelund b6ee3b7685ad2a1197de94902f15beab6a743a6e solved the system library dependencies for MSVC in a platform-specific way? Is there some similar pragma for GCC on MinGW?

sjoelund, 2016-10-24:
@tbeu There is no such pragma for GCC. I assume because library linking order matters for GCC as it uses a single-pass linker.

OMC already searches first OMC system path, then Library/mingw32 (or mingw64), then Library/win32 (or win64), then Library.

tbeu, 2016-10-24:
If OMC searches first OMC system path why does it generate linker errors on the pthread.dll from Library/win32?

bernhard-thiele, 2016-10-24:
@tbeu @vwaurich  @adrpo Just looked with Martin into the issue. There is no "pthread.dll" in the OMC system path. Still `-lpthread` is a standard argument for the linker. Now, it seems, that *if* there is a "pthread.dll" available in the resources directory, it takes it. If it is not available it does something unknown special.

sjoelund, 2016-10-24:
Wouldn't it be nice if Modelica included an option to only link libraries on certain platforms?

[34 more replies]

closed 2019-09-17 by bernhard-thiele