modelica-3rdparty/Modelica_DeviceDrivers #115: LCM support wanted?

tbeu, 2016-04-30:
[LCM](https://github.com/lcm-proj/lcm) is for Lightweight Communications and Marshalling. The communication is based on UDP multicast and alternatively supports logfile or memory queue for testing. The LCM protocol is used in robotics by several research institutes, see http://lcm-proj.github.io.

I tested the LCM C library on Windows. The integration of LCM within Modelica, especially M_DD, would be straightforward and similar as UDP or CAN (with one additional library dependency on lib lcm that would need to be part of the M_DD distribution).

The marshalling part of LCM is not possible to use within a Modelica environment (since it is not a nominal type system). Thus the SerialPackager has to be reused for encoding/decoding of the data buffer. Endianess needs to be considered again (as is the case for UDP already).

@bernhard-thiele Please comment if it's alright for you to have support of the LCM protocol in M_DD. I will provide a PR once I am ready then.

bernhard-thiele, 2016-05-02:
Sorry for the late answer, I was away from email etc.

I just scanned over the website and it sounds interesting. If LCM can be supported in a user convenient manner, it would be nice to have it in M_DD. From a first superficial look, LCM seems a bit similar to the Google protocol buffers (https://github.com/google/protobuf), but probably LCM is more optimized for real-time applications.

How are you going to map the LCM data type generation to Modelica (`lcm-gen ...`)? From above I assume that you would create the packages manually by using the SerialPackager. That seems a bit error-prone for the user. Or do you think of writing code that understands the `*.lcm` message definitions and generates Modelica code? 

Another question is how to map the event based subscriber model to Modelica. It is not obvious to me what the best way of doing this would be. The example at the website shows that `lcm_handle` is called periodically and calls out to a function handler to deal with the message.

In summary: It would be nice to have LCM support, though it is not so clear to me how the "look" and "feel" of such a support in M_DD would be. Fine for me to give it a try and help testing/supporting the code under Linux.

tbeu, 2016-05-02:
The lcmgen step is some kind of optional preprocessing once the protocol is defined. It needs to be done once for each kind of application. I do not think that it is possible to use the data strcutures defined by *.lcm or the lcmgen-generated headers in Modelica. Which means the LCM analysis tools are not really useful for this kind of Modelica applications interfacing LCM.

I was thinking of supporting the generic functions lcm_publish and lcm_subscribe and let the SerialPackager of M_DD do the buffer handling. This indeed might be error-prone for the user. It also might be possible that the SerialPackger writes a *.lcm file for the used protocol once during simulation runtime such that the user get's some feedback what the protocal is.

Instead of sampled communication (like UDP or other blocks of M_DD) the LCM blocks could work with continuous time, i.e. lcm_handle will be called in each solver step. Once a new message is available the SerialPacker will get this data, however it is eventless in sense of Modelica. This would be different to the existing communication blocks of M_DD.

I will try to implement it but it may take same days or weeks. I got drone.io running for the x64 build of liblcm.a but I am still failing with cross-compilation for x64_86, see https://drone.io/github.com/tbeu/lcm/admin. Maybe @dietmarw has an idea why glib:i386 fails on Ubuntu 12.04. You are now admin of this project at drone.io, too.

tbeu, 2016-05-02:
@ashuang FYI.

dietmarw, 2016-05-02:
@tbeu You probably want to take a look at https://circleci.com which provides also support for artifacts AND a more modern building environment.

tbeu, 2016-05-06:
LCM support is now added to new branch lcm. Please check. What's left to do:
* Add linux32 and linux64 binaries. I will do it once I have a better network connection to set it up.
* Adapt some CMake files for e.g. test_MDDLCM and ITI_MDDLCMWrapper. It akes me 3 min to set up the VS files but 3 days for the CMake files. Thus, I do not feel in the mood for it.

tbeu, 2016-05-10:
@bernhard-thiele Did you have some minutes to check the new branch?

bernhard-thiele, 2016-05-10:
Yes, I will check it later today.

bernhard-thiele, 2016-05-11:
Had problems to build the example model (_Examples.TestSerialPackager_LCM_). Building the example model with Dymola on 64 bit Ubuntu 14.04 I needed to use the library dependencies `{"pthread", "glib-2.0", "lcm"}` and create the symlink `ln -s libgthread-2.0.so.0 libgthread-2.0.so` within the directory `/usr/lib/i386-linux-gnu`. By default Dymola builds 32 bit binaries, so we need to ensure that the 32 bit libraries of glib-2.0 and others are installed.

Now the model builds, but does not yet run. I get the error

```
The following error was detected at time: 0
MDDLCM.h: Could not allocate LCM object for network provider "udpm://"
```

Too late and to tired for looking into that now.

tbeu, 2016-05-12:
1) As I wrote in the commit message liblcm on Linux requires libgthread-2.0 and libglib-2.0 as system dependency. I wonder why it worked with libpthread for you. For Windows I compiled lcm.dll w/o explicit dependency on glib/gthread, i.e., the required dependencies are statically included in lcm.dll. It would be nice if we could do the same for Linux. @bernhard-thiele, @dietmarw , @sjoelund Do you have an idea how to not use a shared object but rather include the dependent libs?

2) The run-time error is due to lcm_create returning NULL which is strange. I debugged it (on Windows) many times and could not see any problem with it. The lcm project comes with some tests. You could try these tests of the C interface and check them on Linux.

sjoelund, 2016-05-12:
@tbeu choosing to link XXX statically is just `-Wl,-Bstatic -lXXX -Wl,-Bdynamic`. This won't work in a Library annotation I think, and as far as I know if cmake is used it is a bit tricky to get it right, but it can be done.

bernhard-thiele, 2016-05-12:
@tbeu Maybe my firewall is responsible for the run-time error. Just had a quick test using the project examples which suggests that (it worked after deactivating the firewall). Need to look into this and also should play with the library dependencies if I find some time in the evening.

bernhard-thiele, 2016-05-15:
It is a bit difficult to find time in the moment, it is not forgotten and I hope there is some more time next week.

tbeu, 2016-05-16:
Thanks for the reply. Fine from my side.

tbeu, 2016-06-02:
Any time spent on the topic?

tbeu, 2016-06-23:
I can now confirm that it works with Dymola on Windows.

Did you find out what is the case with Dymola on Linux?

bernhard-thiele, 2016-06-28:
I could get it run on my Ubuntu 14.04 64 bit, but maybe it's not very satisfactory. 

1. Needed to disable my firewall
2. I had to change line 86 of  _MDDLCM.h_ from `lcm->lcm = lcm_create(url);` to `lcm->lcm = lcm_create(NULL);`

After that I could simulate the example and saw that values were received in the `getXXX` blocks.

@tbeu the example in the lcm distribution just uses NULL as argument, but I guess that is not acceptable here?

Preceding to this, I had to do one of the following (though it doesn't matter if one does both). 

### Variant 1: Follow the approach described in a comment above (repeated for convenience)

Use library dependencies `{"pthread", "glib-2.0", "lcm"}` and create a symlink `ln -s libgthread-2.0.so.0 libgthread-2.0.so` within the directory `/usr/lib/i386-linux-gnu`. By default Dymola builds 32 bit binaries, so we need to ensure that the 32 bit libraries of glib-2.0 and others are installed.

### Variant 2: Installation of a 32 bit lcm installation

Since Dymola defaults to 32 bit code generation on Linux one needs to install the 32 bit version of lcm, which can be done by building the source distribution of lcm (v1.3.1 in my case) with

```
./configure --without-python --enable-static --build=i686-pc-linux-gnu "CFLAGS=-m32" "CXXFLAGS=-m32" "LDFLAGS=-m32"
make
sudo make install
```

In this case it is not required to have any lcm linux library shipped with MDD. Furthermore, no additional library dependencies need to be declared in Modelica (`Library="lcm"` is sufficient).

tbeu, 2016-06-28:
Thank you for your time evaluating the lcm branch on Linux.

1. `lcm_create(NULL)` calls a default URL `udpm://239.255.76.67:7667` which - of course - is not generic enough.
1. For Win (both Dymola and SimulationX) I created a lcm.lib linking with lcm.dll that already contains glib. For Linux I'd see the most benefit providing an liblcm.a along with M_DD that also already contains libglib. Shouldn't this be possible. I'd avoid to require any sources of lcm or glib as part of the library.

bernhard-thiele, 2016-06-28:
Well, if you can compile glib for Windows, shouldn't it be possible to simply provide both libraries, "glib.*" and "lcm.*", seperately instead of merging them together?

The advantage would be that

1. We wouldn't introduce a library which is called lcm, but is at the same time glib (hence, we wouldn't introduce something unexpected)
2. We easily could use the services provided by glib in other (future) code, too

tbeu, 2016-06-28:
You are right. I will check it out on Win and update the lcm branch if it worked out.

tbeu, 2016-06-28:
Actually, the gthread lib could replace the pthreads lib then.

tbeu, 2016-06-28:
Right, bb4ad8a provides separate MSVC libs for glib-2.0 and gthread-2.0. The later one actually is not needed, but the LCM project calls it a requirement.

bernhard-thiele, 2016-06-29:
Okay, I'm currently trying to update the cmake build system so that the glib and lcm dependencies can be managed with some convenience.

tbeu, 2016-06-29:
I asked in lcm-proj/lcm#76 about the obsolete dependency on gthread-2.0. I recommend it to remove it again from M_DD. Do you agree?

bernhard-thiele, 2016-06-29:
Yes, it is fine to remove it.

tbeu, 2016-06-29:
Will do so.

tbeu, 2016-07-01:
What are the extra lcm headers are needed for? Why is there sn artificial directory hierarchy lcm/include/lcm when only lcm is enough?

bernhard-thiele, 2016-07-01:
I have added a linux version and extended the cmake system so that it is easy (in Linux) to rebuild the dependencies if needed.

The cmake will download automatically the whole lcm and/or glib in the specified version from Github and setup and after building and installing we have folllowing below lcm:

```
bin  include  lib  share  src  tmp
```

The source zip package is downloaded to `src`. The "installation" copies everything to `lcm`

bernhard-thiele, 2016-07-01:
Currently, this works only in Linux. Of course it would be nice if it would be extended to also generate VisualStudio projekt files for glib and lcm to provide the same functionality under Windows.

bernhard-thiele, 2016-07-01:
What is also missing is a bit of documentation. At least for my Ubuntu I need to do

```
 sudo ifconfig lo multicast
 sudo route add -net 224.0.0.0 netmask 240.0.0.0 dev lo
```

before the LCM examples run and this requirement should be visibly documented.

Do the settings for the default UDP multicast address also work for you under Windows?

tbeu, 2016-07-01:
drone.io reports linker errors. Did you have alook at them?

bernhard-thiele, 2016-07-01:
No, I have not yet looked at it. I will try to understand what is happening.

I still have trouble to fully understand the multicast approach. I can set a multicast address at the sender or at the receiver. But what does that really mean? I just run our example model and the multicast address at the sender and the receiver don't need to be same. Web sites like http://www.tldp.org/HOWTO/Multicast-HOWTO-2.html talk about the need of joining a multicast group in order to receive datagrams. It is currently not clear to me how this relates to the IP addresses that we can give for the sender and the receiver. 

Do you understand that better? Should the documentation and/or the parameter settings be changed?

bernhard-thiele, 2016-07-01:
I don't understand why drone.io still has linker errors. It works on my box. I don't know what to do to fix it.

tbeu, 2016-07-01:
> Do the settings for the default UDP multicast address also work for you under Windows?

Yes, works exactely as before. Thank you!

tbeu, 2016-07-01:
libglib-2.0.a is 7,9MB for x64, but only 1,6MB for x86. Do you have an idea why there is such a large difference in size?

tbeu, 2016-07-02:
Here is my test from OpenSUSE 13.2 32-bit.
* CMake 3.0.2 works as it did for you.
* Manual compilation works:
```
thomas@laptop:~/Modelica_DeviceDrivers/Modelica_DeviceDrivers/Resources/test/Communication> gcc test_MDDLCM.c ../Util/ModelicaUtilities.c -I../../Include -L../../Library/linux32 -llcm -lglib-2.0 -lpthread
```
* Execution fails:
```
thomas@laptop:~/Modelica_DeviceDrivers/Modelica_DeviceDrivers/Resources/test/Communication> ./a.out 
Testing MDDLCM 1.3.1
LCM self test failed!!
Check your routing tables and firewall settings
MDDLCM.h: Could not subscribe callback function to receiver channel "test5"
```
* Receiver does not receive the expected message:
```
thomas@laptop:~/Modelica_DeviceDrivers/Modelica_DeviceDrivers/Resources/test/Communication> sudo ifconfig lo multicast
thomas@laptop:~/Modelica_DeviceDrivers/Modelica_DeviceDrivers/Resources/test/Communication> sudo route add -net 224.0.0.0 netmask 240.0.0.0 dev lo
thomas@laptop:~/Modelica_DeviceDrivers/Modelica_DeviceDrivers/Resources/test/Communication> ./a.out 
Testing MDDLCM 1.3.1
Sent: Current i is 0
Received 
Sent: Current i is 1
Received 
Sent: Current i is 2
Received 
Sent: Current i is 3
Received 
Sent: Current i is 4
Received 
Sent: Current i is 5
Received 
Sent: Current i is 6
Received 
Sent: Current i is 7
Received 
Sent: Current i is 8
Received 
Sent: Current i is 9
Received 
```

tbeu, 2016-07-02:
And here are my investigations on drone.io:
* Note that file build.drone.io.txt is not relevant for drone.io CI. It only is there for remembering the settings from https://drone.io/github.com/modelica/Modelica_DeviceDrivers/admin.
* After updating CMake from 2.8.7 to 3.5.2 on drone.io the build works: https://drone.io/github.com/modelica/Modelica_DeviceDrivers/112
* But cross-compilation still fails with a linker error: https://drone.io/github.com/modelica/Modelica_DeviceDrivers/113
* Thus I removed the 32-bit build on drone.io and can see the empty receiver messages: https://drone.io/github.com/modelica/Modelica_DeviceDrivers/114
* Also need to upgrade CMake also on Travis CI where there is the same CMake issue.

bernhard-thiele, 2016-07-04:
Yes, I think that would be good if you could add me to drone.io and Travis CI. I just got an account at both services by using my Github account.

I also checked regarding the large size difference between the 64 bit libraries vs the 32 bit libraries. 
In the 32 bit case I do `CFLAGS=-m32` to enforce 32 bit compilation. However, this overwrites the default settings `CFLAGS = -g -O2`, and thus no optimized code is generated in the 32 bit case. If I change that to `CFLAGS = -g -O2 -m32` the size of the generated libglib-2.0.a increases to 4.7MB instead of 1.6MB. Inversely, if I set `CFLAGS = -g -O0` in the 64 bit build, libglib-2.0.a will decrease to 3.8MB instead of 7.9MB. Thus, using code optimization adds quite some size to the binary.

bernhard-thiele, 2016-07-04:
I can't reproduce https://drone.io/github.com/modelica/Modelica_DeviceDrivers/114. If I translate the project created by `CFLAGS=-m32 CXXFLAGS=-m32 cmake ..` and go to the test folder I can do:

```
$ file test_MDDUDPSocket 
test_MDDUDPSocket: ELF 32-bit LSB  executable, Intel 80386, version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 2.6.24, BuildID[sha1]=a602a0d762decb45492b72cdf50b889163c16c42, not stripped
$ ./test_MDDUDPSocket 
Created socket handle: 3 (receiving socket)
Binding receiving UDP socket to port 10002 ...
Created socket handle: 4 (sending socket)
Started dedicated UDP receiving thread listening at port 3
Received 0 bytes: 
Received 80 bytes: Current i is 0
Received 80 bytes: Current i is 0
Received 80 bytes: Current i is 1
Received 80 bytes: Current i is 3
Received 80 bytes: Current i is 4
Received 80 bytes: Current i is 5
Received 80 bytes: Current i is 6
Received 80 bytes: Current i is 6
Received 80 bytes: Current i is 7
Closed UDP socket with socket handle 3
Closed UDP socket with socket handle 4
```

tbeu, 2016-07-04:
I've added you both to https://drone.io/github.com/modelica/Modelica_DeviceDrivers and https://drone.io/github.com/tbeu/lcm with admin rights. Please do not delete these accounts or remove me from them.

I recommend the tested settings from https://drone.io/github.com/tbeu/lcm/admin when compiling libs lcm and glib. There is no need to define any debug information in it. This is also the way how I generate the Linux libs of the Modelica Standard Tables for the MSL.

Does test_MDDLCM receive the messages as expected. It does not on my OpenSUSE 13.2 32-bit and on drone.io. I have no problem to accept this behaviour as valid if it works on your machine.

tbeu, 2016-07-04:
For Travis CI you can edit travis.yml. I have no idea if it is possible to add a member to the project.

bernhard-thiele, 2016-07-04:
test_MDDLCM works on my Ubuntu 14.04. Of course it also uses the static lib that I have compiled on the same machine. So I hope that the libs that are distributed are also working on most other machines. On my machine I have:

```
$ gcc -m32 test_MDDLCM.c ../Util/ModelicaUtilities.c -I../../Include -L../../Library/linux32 -llcm -lglib-2.0 -lpthread
$./a.out
Testing MDDLCM 1.3.1
Sent: Current i is 0
Received 
Sent: Current i is 1
Received 
Sent: Current i is 2
Received 
Sent: Current i is 3
Received Current i is 0
Sent: Current i is 4
Received Current i is 1
Sent: Current i is 5
Received Current i is 2
Sent: Current i is 6
Received Current i is 3
Sent: Current i is 7
Received Current i is 4
Sent: Current i is 8
Received Current i is 5
Sent: Current i is 9
Received Current i is 6
```

tbeu, 2016-07-04:
1. Can you please try with the lcm lib (x64 only) compiled on drone.io: https://drone.io/github.com/tbeu/lcm/files
1. Can you please commit the non-debug libs. Thanks.

tbeu, 2016-07-04:
I run test_MDDLCM repeatedly and in approx. 1 out of 20 tries the receiver saw the expected messages. Thus I believe it is a timing issue we no longer need to investigate. Instead can you please care on the other issues: CMake 3, CMake cross-compilation and non-debug libs of lcm and glib-2.0 (preferable from drone.io).

bernhard-thiele, 2016-07-05:
Yes, I should have some time at Thursday to get this done.

bernhard-thiele, 2016-07-07:
> 1. Can you please try with the lcm lib (x64 only) compiled on drone.io: https://drone.io/github.com/tbeu/lcm/files

The lib works on my machine.

tbeu, 2016-07-07:
So the idea would be to compile both glib and lcm from sources (linux32 and linux64) on https://drone.io/github.com/tbeu/lcm?

bernhard-thiele, 2016-07-07:
> So the idea would be to compile both glib and lcm from sources (linux32 and linux64) on https://drone.io/github.com/tbeu/lcm?

I think this is not necessary. We have checked in precompiled versions of the external libraries and we can just use this libraries for our CI purposes. It doesn't provide new insights if we recompile the external libraries whenever we make a change to our code. 

However, it is good if it works in principle. That way we know that the Cmake scripts work at least for one other Linux distro, too :-)

bernhard-thiele, 2016-07-07:
Hm, what's left to do?

tbeu, 2016-07-07:
> So the idea would be to compile both glib and lcm from sources (linux32 and linux64) on https://drone.io/github.com/tbeu/lcm?

Done by https://drone.io/github.com/tbeu/lcm/39. Can you pls check before I commit them!

tbeu, 2016-07-07:
Oops, cross-typing.

> Hm, what's left to do?

 * Commit binaries from drone.io (for reference)
 * Do a selective git squash merge
 * Update drone.io settings after merge
 * Update Travis CI after merge

tbeu, 2016-07-07:
New binaries are here: https://drone.io/github.com/tbeu/lcm/files

bernhard-thiele, 2016-07-07:
> Done by https://drone.io/github.com/tbeu/lcm/39. Can you pls check before I commit them!

Seems you base the compilation on the glib master branch. I would rather use a stable release. I wonder why you don't let the cmake script do the work? Using something like

```
CFLAGS=-m32 CXXFLAGS=-m32 cmake -DBUILD_EXTERNAL_PROJECTS=true ..
```

The cmake script should download the specified releases for glib and lcm (I used the same glib release that also ships with my Ubuntu) and build 32-bit versions of the libraries.

Regarding the compilation parameters that you use, I would stay closer to the defaults selected by configure (I assume that the glib people have tested with that flags and `-O3` sometimes does too much). Hence, rather have `CFLAGS=-g -O2` and `CXXFLAGS=-g -O2` than `-O3`. I don't know if `-fPIC` is useful here. We are creating a static library and I think it has no benefit to have position independent code in that case.

tbeu, 2016-07-07:
I do not know what -fPIC is for but @sjoelund told me that it always is needed for a successful linking.

I prefer the non-debug binaries, i.e. w/o -g and with NDEBUG defined since I simply have no idea why -g should be there.

I am not a cmake expert. I stick with configure whereever I can and was lucky to get it finally working for 32-bit and x64.

Yes, I can build latest glib 2.49.2 instead of head version.

tbeu, 2016-07-07:
Done by https://drone.io/github.com/tbeu/lcm/40, which is released glib 2.49.2 compiled with -O2 and w/o debug.

tbeu, 2016-07-08:
You have admin rights at https://drone.io/github.com/tbeu/lcm thus you can build it using CMake if you like. Just save the current (working) settings beforehand.

bernhard-thiele, 2016-07-08:
I tested the 64-bit build of cmake including the external projects at https://drone.io/github.com/modelica/Modelica_DeviceDrivers/119. The 64-bit build works.

I also tested 32-bit, but that didn't work. Probably some missing 32-bit libraries.

bernhard-thiele, 2016-07-08:
I can't find how to have several drone.io configurations for one github project. Is that possible?

I would like to have one configuration for 64-bit and one for 32-bit, one that builds the external projects, another one that doesn't... Is this possible?

bernhard-thiele, 2016-07-08:
> For reproduction it would be nice to use the builds from drone.io.

I don't see much improvement from that. Since the binaries (be it for Linux of for Windows) are under version control, reproduction of bugs is already facilitated. Maybe it would be nice to have a mechanism that separates the binary handling from the source code handling (so that the source code repo does not blow up in size). But as long as the current approach works reasonably well, I don't want to invest time into that. 

What is much more important to me is that the builds are reproducible and work highly automatized on various platforms. I think the reproducibility of the Linux lcm/glib part it is now in good shape. What is probably not as ready is the Windows part. Any idea how to facilitate reproduction of the Windows builds?

tbeu, 2016-07-09:
> I can't find how to have several drone.io configurations for one github project. Is that possible?

Do not think so.

> I would like to have one configuration for 64-bit and one for 32-bit, one that builds the external projects, another one that doesn't... Is this possible?

Do not think so.

tbeu, 2016-07-09:
Updated required minimum CMake evrsion to v3.0.2 by 11a36be.

[21 more replies]

PR #125 (2016-07-14): Add support for LCM
* See https://github.com/lcm-proj/lcm for Lightweight Communications and Marshalling
* Close #115
closed 2016-07-14 by tbeu