modelica-3rdparty/Modelica_DeviceDrivers #116: USB/Bluetooth support wanted?

tbeu, 2016-05-03:
@bernhard-thiele Collecting more ideas: https://github.com/signal11/hidapi is a platform independet implementation for USB and Bluetooth communication by the HID API. Do you think it might be useful to have it supported with M_DD?

bernhard-thiele, 2016-05-04:
Could be nice to have. Of course it would add additional dependencies and maintenance efforts, hence it would be good to know if there are any interested users or use cases that you have in mind. Alternatively, one could think on using the OS APIs directly, so that no additional third party dependency is added.

vwaurich, 2016-10-20:
I would be very interested. I have to put some effort in this topic soon and it would be nice to know which api you prefer.

tbeu, 2016-10-20:
Thanks for you interest. I'd go for the HID API, because
* it is platform independent already
* we do not have to re-invent the effort
* we have successfully proven to manage third-path libraries (like glib-2.0 and lcm for #115).

vwaurich, 2016-10-20:
I am not an expert on that, but I think hid api offers support for hid bluetooth profile only. If I want to send and reveice data like in a serial port, I need spp profile. hdi-api can be implemented but the use cases are limited to hdi, am I right? https://learn.sparkfun.com/tutorials/bluetooth-basics

tbeu, 2016-10-20:
Good point. If SSP is what you need then HID API is not the API of choice.

vwaurich, 2016-10-28:
I tried to send data via bluetooth to a modelica simulation in dymola under windows.
It did work without any modification at the serial port code because you can handle bluetooth connections like ordinary COM ports under Windows. But it still shows some arbitrary behaviour. It works only every third time. Thats weird. I will investigate this.

tbeu, 2016-10-28:
With your patch applied I guess?

vwaurich, 2016-10-28:
@tbeu Yes, but it should have no effect on the weird behaviour. When its not working properly it doesnt call ReadFile at all. What I would like to state is, that I dont think we need an additional implementation that tackles bluetooth. Sending and receiving should work with the current serial code. Or are there any other applications of interest except SPP?

vwaurich, 2016-10-28:
If I have a working setup, I can also try it on linux.

tbeu, 2016-10-28:
If Bluetooth via SSP is working for you, we can be glad to have less work. HID might not be of that interest. I only learned now.
What about USB?

vwaurich, 2016-10-28:
Oh...I did not mentioned yet,, that I always use the serial port via usb? So this should be done with my patch.

tbeu, 2016-11-01:
Can you give me some clue where is wire and where wireless communication. I cannot imagine right now.

vwaurich, 2016-11-02:
@tbeu Sorry, I dont get the question.

tbeu, 2016-11-02:
Well, bluetooth is wireless communication. Now that you use (wired?) serial I/O I wonder where there still is wireless I/O? Maybe a stupid question. Thus, would be interesting to read about the application  :smirk:.

vwaurich, 2016-11-02:
The application is reading angular sensors with an arduino and transfering them via a serial usb connection. The same task can be accomplished using a hc-05 bluetooth module to use the arduino without usb connection. Some students developed a mobile handheld control device (like a joystick) and therefore it needs to be wireless.

tbeu, 2018-09-19:
I guess it is ok to close since Bluethooth is working via serial port i/o.

closed 2018-09-19 by tbeu