modelica-3rdparty/Modelica_DeviceDrivers #117: SerialPort receiver thread overwrites results

sjoelund, 2016-05-19:
The implementation of the serial port receiver currently has the same behavior as UDP when receiving a `package`. The problem is that every single byte is polled and received individually especially when sent at a lower baud.

It would be nicer if one would send in the number of bytes to read (via a packager) to the read function, and just let the buffer be a buffer that was filled up when receiving messages and cleaned when reading (but only removing the requested number of bytes). Or at the very least if multiple bytes were received over a serial port, that the packages were appended.

bernhard-thiele, 2016-10-25:
I will work on this (related #119, #127)

bernhard-thiele, 2016-10-26:
I will close this issue since #128 should have solved the issue for Linux and the Windows related discussion has started at #127.

closed 2016-10-26 by bernhard-thiele