modelica-3rdparty/Modelica_DeviceDrivers #120: Fresh start of refactoring of Linux code

bernhard-thiele, 2016-05-19:
Commits modelica/Modelica_DeviceDrivers@3cbe5fea91ec6d3fd1dcfb193b1cc8a116fffc1c...1b9459b5d1decff3900f3af1528aaa869641a599
have introduced many changes at once in the Linux code. I don't agree with some of the chances, e.g., the silent null pointer tests (see e.g., comments to https://github.com/modelica/Modelica_DeviceDrivers/commit/1b9459b5d1decff3900f3af1528aaa869641a599).

If there seem to be bugs or other issues in the Linux C code I would rather prefer to discuss them individually before applying a bunch of changes. Some of the code is used in serious applications and I want to ensure that I understand decisions and consequences properly and that reasonable testing is done before pushing changes. For that reason I will revert the last changes to allow a fresh start.

bernhard-thiele, 2016-05-19:
@tbeu Pushed reverting commits 30ed182f9420630d85a921435a83638eefbc6c08...d354b881bf17bf013b695e7eb683cb3cdea17132

tbeu, 2016-05-20:
Why is this issue closed?

bernhard-thiele, 2016-05-23:
I thought of handling each issue in its own ticket. However, it could be convenient to collect the discussion at one place. I'm fine with reopening the ticket.

bernhard-thiele, 2016-11-16:
Looks like this can be closed.

tbeu, 2016-11-16:
Yes, it basically is now back in shape as it was with my commits 3cbe5fea91ec6d3fd1dcfb193b1cc8a116fffc1c, 506ec7cf304115e1329a7b97a42c965f509039e9, fdad81af1e3914045bec7cdba848b78afb0baac9 and 02443d89767ea85bd84f1bc4ca9b98fb72a1ec4c (except the consistent naming issue #150 which I excluded for issue 120.).

I still would put https://github.com/modelica/Modelica_DeviceDrivers/blob/master/Modelica_DeviceDrivers/Resources/Include/MDDSerialPort.h#L719 and https://github.com/modelica/Modelica_DeviceDrivers/blob/master/Modelica_DeviceDrivers/Resources/Include/MDDUDPSocket.h#L659 in an else branch.

tbeu, 2016-11-17:
Reopen for the sake of my last comment (and it will not be forgetten).

bernhard-thiele, 2016-11-17:
"Back in shape"? Honestly, this was a good deal of nit-picking of issues that are highly unlikely to get in way of any user. Anyway, I have fixed them in a way that I still can feel confident that the resulting code will not bring in new problems which hit people with real applications, while solving issues that never before have fallen on anybodies feet.

This issue is closed.

closed 2016-11-17 by bernhard-thiele

tbeu, 2016-11-17:
Nit-picking or back in shape - just different views for doing maintenance work, i.e. fixing bugs in software.

tbeu, 2016-11-17:
It's same about malloc/calloc. It succeeds in almost every call, still it can fail and developers need to care about error handling. For you it is nit-picking, for me it is fool-proof.

bernhard-thiele, 2016-11-17:
Your "back in shape" claim suggests severe problems, there only small issue were involved, and together with some other comments have a somewhat derogatory undertone. This might be your opinion, fine, I don't share it. However, the issue tracker is not a good place to elaborate on it and I hope we can finish here.