modelica-3rdparty/Modelica_DeviceDrivers #122: Structurally singular system

tbeu, 2016-06-23:
Reusing a (serial) package for multiple senders results in a structurally singular system.

[image]

bernhard-thiele, 2016-06-27:
The `PackageIn` and `PackageOut` connectors have both, inputs and outputs.
Hence, there is also a backward signal flow and this is why the example above doesn't work.

tbeu, 2016-06-27:
Is it a current limitation of the SerialPackager we need to accept or is there a workaround. I consider this limitation as rather critical.

bernhard-thiele, 2016-06-27:
This is a limitation and I don't see a simple workaround. One could provide a distinguished splitter block, but one needs think how to handle the back propagation. Also, once one can split into parallel branches, somebody could be tempted to add packages in parallel branches.

Another workaround could be to have a feedthrough at a send block which allows to connect another send block in series. Not sure what is more feasible.

tbeu, 2016-06-27:
Since package branching is not allowed anywhere, I would prefer a splitter block.

Do you know of any (tool-specific) way to prohibit branching, i.e. have max (exactely) 1 connection per connector? We could use cardinality (like in MSL Fluid) but it is deprecated.

tbeu, 2016-06-27:
https://trac.modelica.org/Modelica/ticket/1410 proposes to use cardianality only in asserts. Thus we could simply do that.

bernhard-thiele, 2016-06-27:
I have no idea how one could prohibit this and I also have no clear picture how to implement. Probably it needs a trial and error approach.

tbeu, 2016-06-27:
Adding an `assert(cardinality(..))` for each element with PackageIn and PackageOut connectors could help. What do you think?

bernhard-thiele, 2016-06-27:
For me it seems that in this way one maybe could get better error messages for systems that branch out, but it still wouldn't allow you to have several sinks as you would like to have (or am I missing something)?

tbeu, 2016-06-27:
Yes, instead of the singular sysem error msg you would get the cardinality assert.

Parallel branches are simple not possible. I beleive a user needs to define its own block if he/she wants to model multiple sinks.

tbeu, 2016-06-29:
Too bad, adding a cardinality assert message does not help the user since the error message on structurally singularity is raised before the assert (in SimulationX).