modelica-3rdparty/Modelica_DeviceDrivers #127: Fallback for serial port if message is incomplete

vwaurich, 2016-10-21:
If you use serial port to communicate to external hardware, you have to ensure send and receive is synchronized. If both sample frequencies do not match very precisely, the received message might be incomplete every n samples and you get a strange step in the results. I have added to the win implementation a fallback buffer, that stores the last complete message to get rid of these steps.
If you are fine with that, I can have a look at the linux version, too.
Have a look at:
https://github.com/vwaurich/Modelica_DeviceDrivers/commit/12a9b7bda942a20894f83efafcf722f1b2ebc271

tbeu, 2016-10-21:
You forgot to free the backup buffer in MDD_serialPortDestructor.

tbeu, 2016-10-21:
`ret = false;` is not ANSI C but C++ style. Correct is `ret = FALSE;`

tbeu, 2016-10-21:
What is the return value of ReadFile in `case serial->receivedBytes != serial->bufferSize`? Should you check `serial->receivedBytes < serial->bufferSize` actually?

tbeu, 2016-10-21:
FYI: http://www.codeproject.com/Articles/81933/Serial-Port-R-W-With-Read-Thread was the source for serial port communication on Win.

tbeu, 2016-10-25:
Same question here (w/o acceptable answer though): http://stackoverflow.com/questions/30950132/incomplete-reading-data-serial-port
This one is more interesting: http://forums.codeguru.com/showthread.php?298713-ReadFile-not-reading-in-values-fast-enough-from-Serial-Port

sjoelund, 2016-10-25:
I guess this is related to #117 as well.

tbeu, 2016-10-25:
Further improvements on MDDSerialPort.h (before I forget)
* No need to have receiveError in struct MDDSerialPort
* ClearCommError checks for unsupoorted CE_IOE flag

tbeu, 2016-10-25:
> I guess this is related to #117 as well.

Possibly. But this case is Win specific (ReadFile issue).

tbeu, 2016-10-25:
The Win implementation of serial port I/O is asynchronous (overlapped). What about the Linux implementation?

tbeu, 2016-10-25:
This is Ranga's first reply:

> Hi Thomas, 

> I've just now seen the thread and also looked at your suggestions.
My suggestions:
A. The front end reads the buffer, but doesn't have a read complete flag or indication. Or a clear operation which might cause a superfluous re-read.
B. In my application the data coming in to the Serial buffer was always sent by an Embedded system, so the requests and responses were always in sync.
C.I am not 100% sure the persons (who has raised the issue)application does both A. Or B.?
D.Few variables in the Read Thread usage can be turned into Volatile so that there is no optimization if at all that's the cause of the failure.

> I will analyze further over the weekend, but do give me a flow of how it is used between Serial Port and front end. I feel that is the root cause at least the point of cause.

> BR
Ranga

bernhard-thiele, 2016-10-26:
@vwaurich The issue should be fixed for linux 3c0a9da5b5d782b6291145a03ebefc8b860105c6.

Are you interested to fix it for Windows?

tbeu, 2016-10-26:
Let's wait for Ranga's next reply and see if he comes up with some good idea.

vwaurich, 2016-10-27:
@bernhard-thiele @tbeu There is a PR with fixes for the serialPort win implementation. For my use cases, SerialPort is now applicable.

tbeu, 2016-10-27:
Nice!

tbeu, 2016-11-14:
This ticket can be closed.

bernhard-thiele, 2016-11-16:
Since the PR is merged, I suppose I can close this issue.

closed 2016-11-16 by bernhard-thiele