modelica-3rdparty/Modelica_DeviceDrivers #129: Remove separate getReceivedBytes functions from communication drivers

tbeu, 2016-10-25:
Separate issue split from #128: I propose to completely remove MDD_serialPortGetReceivedBytes (and all other getReceivedBytes) functions and instead return the read buffer size in MDD_serialPortRead**P** (and all other read package functions) and make it available as result of the communication block. Only, this way we can be sure that read buffer size is consistent between function calls. Doing so we can also get rid of the mutex on serial.receivedBytes (Win) or serial.nReceivedBytes (Linux) (and similar variables in other communication structs). Note, that DD_serialPortRead (and all other read functions) are obsolete and are not used throughout the library.

tbeu, 2016-10-25:
@bernhard-thiele Please assign to me or you, depending on your choice and opinion.

bernhard-thiele, 2016-10-26:
As far I see it the two functions we talk about are

- UDPSocket: getReceivedBytes(..) and
- SerialPort: getReceivedBytes(..)

I don't want to remove them for compatibility reason. Flagging them as "deprecated" should be enough.

Not sure whether returning the read buffer size in MDD_serialPortReadP would be intuitive to interpret or help a user. If it doesn't help or is rather confusing we don't need to provide it.

For example, if a message is only partially transmitted, MDD_serialPortReadP would return the number of read bytes. It would not necessarily update the package (it sticks with the last complete message until the next message is complete). However, whether the last read bytes have lead to a new complete package or not remains unclear. Hence, the information that is probably most interesting (whether we have an updated package or not) is not revealed.

tbeu, 2016-10-26:
In SharedMemory it is called getDataSize, thus there is one more.

Yes, we could extend them from Modelica.Icons.ObsoleteModel to mark them as deprecated.

sjoelund, 2016-10-26:
> Yes, we could extend them from Modelica.Icons.ObsoleteModel to mark them as deprecated and leave everything else as it is. This would be fine for me.

Can you? Aren't they functions?

tbeu, 2016-10-26:
Works in SimulationX and Dymola:
```mo
model Unnamed
  model A
    function f
      extends Modelica.Icons.Function;
      extends Modelica.Icons.ObsoleteModel;
    end f;
  end A;
  A a annotation (Placement(transformation(extent={{-20,0},{0,20}})));
  annotation(uses(Modelica(version="3.2.1")));
end Unnamed;
```

bernhard-thiele, 2016-10-26:
SharedMemory is a bit different since getDataSize just returns the size of the shared memory area, not what was read. This value is set by the constructor, so it is much like a parameter value.

Regarding my total bytes read implementation in #128, probably I shouldn't do that either. I will change it again. It would be (more) useful to have such a total value, but from a strict compatibility view, one should rather introduce a new function instead of "redefining" the buggy one.

tbeu, 2016-10-26:
> but from a strict compatibility view, one should rather introduce a new function instead of "redefining" the buggy one.

Agreed.

bernhard-thiele, 2016-10-26:
Just committed 5ce67508e76e4a6bd8519b61ab7bcf28ab37cfe2 which deprecates the SerialPort getReceivedBytes. I will do something similar for UDPSocket.

@sjoelund @tbeu Do you agree that #128 can be merged now, or do you so more crucial issues?

sjoelund, 2016-10-26:
A lack of commit squashing? I should probably test it on our board as well..

tbeu, 2016-10-26:
> Do you agree that #128 can be merged now, or do you see more crucial issues?

Will check now, hang on!

tbeu, 2016-10-26:
> A lack of commit squashing?

Can do squashed cherry-picking as well.

tbeu, 2016-10-26:
* Do you want to update the test by not using the deprecated functions?
* Comment `"DEPRECATED. Get  number of received bytes"` has one superfluous blank after `Get`.
* ``"total number of received bytes"` still has the `total` here, which does not belong there.
* `@deprecated Returns the number bytes`: Missing `of`.
* Is `msgLastComplete` still there by design or side-effect? This belongs to #128, right?

bernhard-thiele, 2016-10-26:
Thanks. I correct these, but I didn't find the line `"total number of received bytes"`. 

The `msgLastComplete` is by design. `msgInternal` is used as buffer for the `read(..)` system call. It may contain partial reads, and thus partial messages. Once a complete message has been read into `msgInternal` the buffer is copied into the buffer `msgLastComplete`. The Modelica side will only deal with `msgLastComplete`.

I can squash the commits to beautify the history. But before, I will first check with @sjoelund whether the new code works with our test board.

bernhard-thiele, 2016-10-26:
After another bug was fixed (see fb37ae0c317b13e92cb045d92b781377fd5a9ae0) we managed to set values and receive sensor values over the serial port for this board: http://vlabs.iitb.ac.in/sbhs/.

I don't see the need to take the deprecated functions out of the tests. It it is available, it is fine to have it in the test suite. If we remove it some day, it is no problem to adapt the tests.

I will squash the commits

bernhard-thiele, 2016-10-26:
I just started #131 which uses the obsolete marker for the respective functions. As I understood @tbeu you didn't like that the icon is just a red frame. What I did was to use the standard function icon from the Modelica.Icons.Function to display the MSL function icon for functions. This looks like
[image]

We can do that consistently for the whole library. What do you think?

bernhard-thiele, 2016-10-27:
The respective functions have been marked with the "ObsoleteModel" icon and all functions consistently use the "Function" icon (see #131, 6d6be5d9197c0cf707ef846924f47746cceac888). Hence, I think this issue can be closed.

tbeu, 2016-10-28:
@bernhard-thiele The functions in Blocks and ClockedBlocks did not get the Modelica function icon. Did you forget them or skipped them by design?

tbeu, 2016-11-01:
5007b4f87cfba320a5968448670fb9b9686d22df removed the (by then) unused obsolete icon. I propose to revert that commit.

tbeu, 2016-11-08:
[mo:#2100](https://trac.modelica.org/Modelica/tickets/2100) also effects this issue/library.

tbeu, 2018-09-19:
This was forgotten to close.

closed 2018-09-19 by tbeu