modelica-3rdparty/Modelica_DeviceDrivers #130: MQTT client

tbeu, 2016-10-25:
See [Wikipedia](https://en.wikipedia.org/wiki/MQTT) and [mqtt.org](http://mqtt.org/) for an description of  the Message Queuing Telemetry Transport protocol and its usage for the Internet of Things.

Based on the experience with LCM and the TCP/IP and by wrapping the platform-generic [Eclipse Paho MQTT C client](https://github.com/eclipse/paho.mqtt.c) I am going to implement an MQTT client for both asynchronous and synchronous communication. Development branch is `mqtt`.

@vwaurich FYI

tbeu, 2016-12-20:
@vwaurich @bernhard-thiele I already started a client implementation. I guess it will be finsihed after the article submission and before the next release / Modelica conference.

Implementation is straith-forward actually, similar to LCM. Pthread dependency is only needed on Linus, but not in Win.

PMehrfeld, 2018-07-02:
I just noticed this interesting approach for implementing a MQTT solution for Modelica. Is this still planned for v1.6.0. And if so, does already exist an official release date?

Thanks a lot in advance.

bernhard-thiele, 2018-07-02:
There exists first prototypical work by @tbeu, but I don't know whether that has matured or whether it is on hold.

There is no planned release date for v1.6.0, yet. It will be released as soon as there are some suitably mature new features in the master branch. This is currently not the case (but I agree that it would be nice to have it in a not too far away future).

tbeu, 2018-08-13:
@PMehrfeld I restarted work on the MQTT client solution for Modelica. Do you already know if you require SSL support? I'd skip it otherwise in the first implementation.

PMehrfeld, 2018-08-14:
@tbeu : Glad to here that you are proceeding the work on the MQTT solution. No, we do not need SSL support.

tstorek, 2018-08-14:
@tbeu thank you in advance. If possible ssl support would be an interesting feature for some use cases.

tbeu, 2018-08-15:
@tstorek Good news, I've got atll the environment set up and managed to connect with [test.mosquitto.org](http://test.mosquitto.org) using ports 1883 (MQTT, unencrypted), 8883 (MQTT, encrypted) and 8884 (MQTT, encrypted, client certificate required) (on Windows, based on the simple test in https://github.com/eclipse/paho.mqtt.c/issues/240). Support of websockets is currently in development for [paho.mqtt.c](https://github.com/eclipse/paho.mqtt.c), but not yet released. Let me know, if you required it already now, but I'd rather prefer to stay at [v1.2.1](https://github.com/eclipse/paho.mqtt.c/releases/tag/v1.2.1), which is the latest release.

tbeu, 2018-08-20:
Status update: I successfully tested paho.mqtt.c with websocket support, which will be released in a few weeks as v1.3.0.

PR #256 (2018-09-26): Add support for MQTT client
Tested with MSVC and GCC on Linux.

TODO:
- [x] Add ITI_MDDMQTT.dll to the Win libraries
- [x] Wait for official paho-mqtt.c v1.3.0 release (especially with https://github.com/eclipse/paho.mqtt.c/pull/521 merged) and update libs + headers
- [x] The clientID should be shorter (max. 23 characters, typically) and could depend on getMACAddress instead of getInstanceName. Need to wait for #255 getting merged.
- [x] Build OpenSSL with no-dso to avoid linker dependency on libdl

I did not spent time on setting up a local MQTT broker, thus the example works on some public t [... cut]
commit 057b9798 (2018-09-26): Add support for MQTT client (#256)

Add support for MQTT client

Closes #130.

Below follows the list of commit messages squashed into this single commit.

* Add third-party headers for Eclipse Paho MQTT C client

* Update Readme.txt to include Eclipse Paho MQTT C client

* Add static libraries of Eclipse Paho MQTT C client for Linux

Built at drone.io (https://drone.io/github.com/tbeu/paho.mqtt.c/files) with custom Makefile:

```
AR = ar -ru
RM = rm -f

CFLAGS = -O3
CPPFLAGS = -DNDEBUG
INC =

TARGETDIR = linux64

# The names of the four different libraries to be built
MQTTLIB_C = paho-mqtt3c
 [... cut]
closed 2018-09-26 by bernhard-thiele