modelica-3rdparty/Modelica_DeviceDrivers #141: Improve memory allcoation handling in socket CAN receiver (Linux)

tbeu, 2016-11-06:
ModelicaAllocateString is called in MDD_socketCANRead, which terminates the complete simulation in case of a memory allocation error. The recommended workaround is to call ModelicaAllocateStringWithErrorReturn instead with proper error handling.

See #120 for the general issue.

sjoelund, 2016-11-07:
Note: The simulation is not terminated if MdoelicaAllocateString is called. It is the same as a failed Modelica assert() or calling ModelicaError. The simulation should be able to recover from this.

tbeu, 2016-11-07:
I beleive in case of SimulationX the simulation stops at ModelicaError. And the [default implementation](https://github.com/modelica/Modelica_DeviceDrivers/blob/master/Modelica_DeviceDrivers/Resources/test/Util/ModelicaUtilities.c#L35-L39) of ModelicaAllocateString is to call ModelicaError in case of memory allocation error. The new ModelicaWarning is thought to be similar as Modelica assert().

sjoelund, 2016-11-07:
You are wrong. ModelicaWarning is the same as the warning assertion level. And ModelicaError is the same as the (default) error assertion level. At least Dymola and OpenModelica will try with a shorter step size when an assertion triggers (to see if the assertion can be avoided by reducing numerical errors).

tbeu, 2016-11-07:
Thanks for the clarification. Do you have a nice little example, that I could use to debug and check for ModelicaError? Unfortunately there is none in https://github.com/modelica-compliance/compliance and the [one example](https://github.com/modelica-compliance/compliance/issues/10) that I found is not suitable.

sjoelund, 2016-11-07:
I guess there should be examples in there using assert. It should be possible to take any of them and replace assert with an external function calling ModelicaError instead.

However, ModelicaCompliance models should work with pretty much any numerical solver so I wonder if retries are actually good for it. I suppose we could have an external function that fails only the first time you call it and put it as part of an algebraic loop?

tbeu, 2016-11-07:
From SimulationX I know that assert with default error level behaves as expected. But I am not sure for ModelicaError right now.

> I suppose we could have an external function that fails only the first time you call it and put it as part of an algebraic loop?

Yes, exactely.

tbeu, 2016-11-07:
Documentation of Modelica.Utilities.Streams.error = ModelicaError says: "Print ... error message and cancel all actions."

sjoelund, 2016-11-07:
The actual Modelica Spec says:
> Output the error message string (no format control). This function never returns to the calling function, but handles the error similarly to an assert in the Modelica code.

tbeu, 2016-11-07:
Could be improved to: "... but handles the error similarly to an assert with (default) error level in the Modelica code."

tbeu, 2016-11-07:
Ah, upcoming MLS 3.4 has already:

> The error functions never returns to the calling function, but handles the error similarly to an assert with assertionLevel=AssertionLevel.Error in the Modelica code.

tbeu, 2016-11-07:
OK, it is a bug in SimulationX: The test model [https://gist.github.com/tbeu/c88491301aded27fe00600cc56b2643d](ModelicaErrorOnce) fails.

tbeu, 2016-11-07:
Cross reference: https://github.com/modelica-compliance/compliance/pull/19

bernhard-thiele, 2016-11-16:
I suppose this ticket can be closed? Reopen, if necessary.

closed 2016-11-16 by bernhard-thiele

tbeu, 2016-11-16:
OK for me. There also is no memory leak if ModelicaAllocateString fails.

Thanks for discussing and clarifying it.