modelica-3rdparty/Modelica_DeviceDrivers #146: Unused mutex for serial port / UDP sender (Linux)

tbeu, 2016-11-06:
MDD_serialPortConstructor / MDD_udpConstructor always initializes the messageMutex where it actually is only needed for the receiver.

See #120 for the general issue.

tbeu, 2016-11-16:
Sorry, to open again, but ticket is on both MDD_udpConstructor and MDD_serialPortConstructor.

closed 2016-11-16 by bernhard-thiele

bernhard-thiele, 2016-11-16:
Not that this changes really matter, it's small code beautifications at best...

tbeu, 2016-11-16:
Do not understand ... You moved the `pthread_mutex_init` to the `if (receiver)` block. This is more than cosmetics, but what this ticket actually is about.

bernhard-thiele, 2016-11-16:
It is now only initialized for receiving sockets. Before that it was always initialized, which didn't do any harm either.

tbeu, 2016-11-16:
So I'd call it a semantical change rather than a cosmetical one. This issue was about unsued mutex.

bernhard-thiele, 2016-11-16:
I've just seen #144 and the simplest way to cater for it seems to always initialize the mutex and always destroy it. That minimizes conditionals and keeps the code simple.

tbeu, 2016-11-16:
All Linux issues are connected to #120 and also linked from there.

commit 8f31e16e (2016-11-16): Small beautification of code. Closes #146
commit f9364e2b (2016-11-16): Small beautification of code for serial port

Closes #146
commit 938c5819 (2016-11-16): Revert "Small beautification of code. Closes #146"

This reverts commit 8f31e16e895ef34728638d4e7cf8d0257369015f.