modelica-3rdparty/Modelica_DeviceDrivers #153: Should external C functions check their external object pointer for non-null?

tbeu, 2016-11-07:
There was an open discussion if the pointer of the external object passed to external C functions should be checked for non-null. This is not consistently considered troughout the library. We already figured out the three options

1. Do not check pointers.
  * Advantage: Short and clear code
  * Disadvantage: Apllication may crash with null-pointer access violation.
2. Silently accept if the pointer is non-null.
  * Advantage: Application no longer crashes with null-pointer access violation.
  * Disadvantage: There might be some kind of undefined behavior in case of a null-pointer.
3. Do error handling if pointer is null.
  * Advantage: Application no longer crashes with null-pointer access violation and no undefined behavior.
  * Disadvantage: Long code and not clear yet, if really necessary.

I am not sure if this actually is an MA issue. See [mo:](https://trac.modelica.org/Modelica/ticket/1436) for a similar issue on character pointer arguments. The question on external object pointers was not yet asked and answered there.

bernhard-thiele, 2016-11-12:
I opt for 1.

One needs to trust that the pointer that the Modelica tool passes into the external C actually points to the expected external object. In general, if the Modelica tools passes in invalid pointers one is out of luck. Only in the special case, that a null pointer is passed, one has a chance to detect the error.  Therefore, I see a valid pointer as a precondition which needs to be guaranteed/checked by the Modelica tool, rather than by the external C code. Hence, my preference for trade-off 1.

tbeu, 2016-11-14:
Since it is different for other external objectes provided by the MA, I am going to ask at the Modelica Trac.

bernhard-thiele, 2019-09-17:
Seems we now have a mix of 1. and 2. at different places in the code.  Anyway, there are pros and cons for the different options and depending on what the code does one might lean towards a different option. For example, if one has a tool which (wrongly) calls a function with a null-pointer in some cases, but these cases are generally save to ignore, it is a practical approach to simply ignore it. Hence, I don't mind if this is not handled 100% the same way throughout the library.

closed 2019-09-17 by bernhard-thiele