modelica-3rdparty/Modelica_DeviceDrivers #164: Static or import library MDDSpaceMouse.lib (MSVC)

tbeu, 2016-11-28:
I wondered about the different sizes of MDDSpaceMouse.lib:
```
win32\MDDSpaceMouse.dll	8.704
win64\MDDSpaceMouse.dll	10.240
win32\MDDSpaceMouse.lib	168.180
win64\MDDSpaceMouse.lib	1.916
```
When building TestInputSpaceMouse with VS as 32-bit application I can see that the binary depends on HID.DLL and SETUPAPI.DLL. Thus win32\MDDSpaceMouse.lib is a static library and MDDSpaceMouse.dll is not required.

When building TestInputSpaceMouse with VS as 64-bit application I can see that the binary depends on MDDSpaceMouse.dll (which then depends on HID.DLL and SETUPAPI.DLL). Thus win64\MDDSpaceMouse.lib is an import library forcing a run-time dependency on MDDSpaceMouse.dll.

Of course, this is not a major problem, but only a minor inconsistency. I have no strong opinion if we should go for a static or import library MDDSpaceMouse.lib since linking worked from VS2005 to VS2015 with both win32/win64.

See also #21.

bernhard-thiele, 2016-11-28:
My intention was to use a static library MDDSpaceMouse.lib which would combine hid.lib and setupapi.lib in one "convenience" library. But I couldn't define this properly in cmake and used a hackish solution which involved printing a message that asks to add the hid.lib and setupapi.lib manually to the generated VS-project before doing the build. 

I think the reason I wanted to use a static lib instead of a DLL, was that the generated stand-alone simulation executable could be moved around more easily, since one does not need to think on including the DLL.

I would like to fix the win64 build to be static, and remove the superfluous DLLs. Unfortunately I can't test with the space mouse hardware, since we don't have one here. Do you have hardware to test the result?

tbeu, 2016-11-29:
I have this [SpaceNavigator®](http://www.3dconnexion.de/products/spacemouse/spacenavigator.html) available as device at office.

bernhard-thiele, 2016-11-29:
Okay, I made a static lib instead of a DLL and removed the MDDSpaceMouse.dlls. Could you please check if it is working with your space mouse?

tbeu, 2016-11-29:
The new win64\MDDSpaceMouse.lib worked when running a code-exported stand-alone model of TestInputSpaceMouse together with the real device. But
* Can you please also compile win64\MDDSpaceMouse.lib with VS2010 like all other MSVC binaries we have in the repository.
* SimulationX builtin simulation requires a MDDSpaceMouse.dll, which now longer is found. Thus the DLL binaries need to be there. For clarification we can make them a ITI_MDDSpaceMouse.dll, but actually since they do not depend on SimulationX-specific ModelicaExternalC.lib, we do not have to do the renaming and just add them back.
* You did not touch the CMake files, right?

bernhard-thiele, 2016-11-29:
- Okay, I will install VS2010 and will do a rebuild.
- We can just add them back and see how it goes.
- I didn't touch the CMake files except for enabling the space mouse target which is commented out by default (`# add_subdirectory(src/SpaceMouse)`) in the top-level CMakeLists.txt.

tbeu, 2016-11-29:
Yes, adding the DLLs back will solve the problem for SimulationX. Sorry, I did not remember until I tested it.

bernhard-thiele, 2016-11-29:
I had some trouble to install VS2010 and associated packages and updates on my machine, but got it working finally. The library in the latest push was generated by VS2010. Please let me know if it works for you.

tbeu, 2016-11-30:
I am afraid the new win64\MDDSpaceMouse.lib from ec1a2d8f4a039375ea60b803fec1cdccde6718a5 is not built correctly, as the dependency on HID.DLL is missing.
```
   Bibliothek "TestInputSpaceMouse.lib" und Objekt "TestInputSpaceMouse.exp" werden erstellt.
MDDSpaceMouse.lib(MDDSpaceMouse.obj) : error LNK2019: Verweis auf nicht aufgelöstes externes Symbol "HidD_FreePreparsedData" in Funktion "MDD_spaceMouseGetDataThread".
MDDSpaceMouse.lib(MDDSpaceMouse.obj) : error LNK2019: Verweis auf nicht aufgelöstes externes Symbol "HidP_GetCaps" in Funktion "MDD_spaceMouseGetDataThread".
MDDSpaceMouse.lib(MDDSpaceMouse.obj) : error LNK2019: Verweis auf nicht aufgelöstes externes Symbol "HidD_GetPreparsedData" in Funktion "MDD_spaceMouseGetDataThread".
MDDSpaceMouse.lib(MDDSpaceMouse.obj) : error LNK2019: Verweis auf nicht aufgelöstes externes Symbol "HidD_GetHidGuid" in Funktion "MDD_spaceMouseGetDataThread".
TestInputSpaceMouse.exe : fatal error LNK1120: 4 nicht aufgelöste externe Verweise.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\bin\x86_amd64\link.EXE"": Rückgabe-Code "0x460"
```

bernhard-thiele, 2016-11-30:
Sorry, no idea why, but hid.lib was missing in the dependencies of the MDDSpaceMouse.lib which I finally committed (didn't show up when I compiled the example, because the example declared itself a dependency to hid.lib). It should be fixed with the previous commit.

tbeu, 2016-11-30:
Confirmed, 83340dfc5beeec51fe8de007470dde165d95cadb solves the issue. Ready to merge. Thank you!

By the way, rewriting the git history always causes conflicts at other users pulling the repo. Luckily I figured it out pretty soon.

bernhard-thiele, 2016-12-01:
Okay, thanks for testing!

It's standard for us to use rewriting or squashing for pull request, but maybe safer if only used for branches on forked repos, rather than branches on the main repository.

closed 2016-12-01 by bernhard-thiele

tbeu, 2016-12-01:
OK, good to know. Never met rewriting for PRs so far, as you can still do the squash at branch merge, even in GitHub web now.