modelica-3rdparty/Modelica_DeviceDrivers #175: Decomposition in subclocks failed.

thorade, 2016-12-06:
This is not an issue, but rather a question: 
When I try to connect `assignClock.y` to any of my models, as input, I get the error message shown in the title. I guess I am using it wrong, but by looking at the examples, also in Modelica_Synchronous, I could not figure out how to do it correctly. Maybe you could extend one of your examples to show the correct usage? Thanks.

```Modelica
within ;
model assign_integrate
  extends Modelica_DeviceDrivers.ClockedBlocks.Examples.TestSerialPackager_UDP;
  Modelica.Blocks.Continuous.Integrator integrator
    annotation (Placement(transformation(extent={{74,40},{94,60}})));
equation
  connect(integrator.u, assignClock1.y[3]) annotation (Line(points={{72,50},{70,
          50},{70,30.4},{64.6,30.4}}, color={0,0,127}));
  annotation (uses(
      Modelica_DeviceDrivers(version="1.4.4"),
      Modelica(version="3.2.2"),
      Modelica_Synchronous(version="0.92.1")));
end assign_integrate;
```

bernhard-thiele, 2016-12-08:
You connect a continuous-time integrator to a clocked discrete-time block. An explicit conversion is needed. Try placing a `Modelica_Synchronous.RealSignals.Sampler.Hold` block between the two.

I will add an example.

closed 2016-12-08 by bernhard-thiele

thorade, 2016-12-08:
Thanks for the explanation and example!

commit a9cade34 (2016-12-08): Extended example as requested in #175

Closes #175