modelica-3rdparty/Modelica_DeviceDrivers #177: InternalPackage icon

tbeu, 2016-12-07:
Should standard icon Modelica.Icons.InternalPackage be prefererd over custom icon Modelica_DeviceDrivers.Utilities.Icons.InternalPackage?

bernhard-thiele, 2016-12-08:
MDD already uses Modelica.Icons.InternalPackage at some places and we should prefer this over the internal icon.

tbeu, 2016-12-08:
Some were forgotten to change.
```
There are errors.
Base class "Modelica_DeviceDrivers.Utilities.Icons.InternalPackage" of model "Modelica_DeviceDrivers.Blocks.Packaging.SerialPackager.Internal" is unknown.
There are errors.
Base class "Modelica_DeviceDrivers.Utilities.Icons.InternalPackage" of model "Modelica_DeviceDrivers.Blocks.Packaging.SerialPackager.Internal.DummyFunctions" is unknown.
There are errors.
Base class "Modelica_DeviceDrivers.Utilities.Icons.InternalPackage" of model "Modelica_DeviceDrivers.Blocks.Communication.Internal.DummyFunctions" is unknown.
```

bernhard-thiele, 2016-12-08:
Ah, changed these as well, but seems I didn't save it. I will fix it.

closed 2016-12-08 by bernhard-thiele

tbeu, 2016-12-08:
Thanks. Fix confirmed.

commit eff977a0 (2016-12-08): Use InternalPackage icon from MSL

Closes #177
commit 2351bceb (2016-12-08): Fixing previous icon replacements commit.

Fixes #177