modelica-3rdparty/Modelica_DeviceDrivers #184: Update third-party LCM 1.3 lib

tbeu, 2016-12-26:
Commit https://github.com/lcm-proj/lcm/commit/9b97f4ef8117b64a0fced54c9e4817f814a2d47e fixes a resourece leak for the LCM 1.3 branch. How to proceed? I remember that I built the MSVC binaries with VS2010 and you with CMake on Linux, right?

bernhard-thiele, 2017-05-11:
I tried the  v1.3-lts branch of lcm, unfortunately it doesn't compile straight away for me. It's probably nothing too big, but I can't spent additional time for it now. I will merge your windows binaries, but keep the Linux on the last official release

tbeu, 2017-05-11:
I can try to build the Linux libs.

tbeu, 2017-05-11:
Sorry, I lost the build config from drone.io. So, let's keep the Linux binaries for M_DD v1.5.0.

closed 2017-05-11 by tbeu