modelica-3rdparty/Modelica_DeviceDrivers #186: Microcontroller icon text and font size

tbeu, 2016-12-26:
The font (size) in icon Modelica_DeviceDrivers.EmbeddedTargets.AVR.Blocks.Microcontroller is intrepreted differently in three tested tools:

Dymola 2017 FD01
[image]

MWorks Sysplorer 2016
[image]

SimulationX 3.8
[image]

Not sure what the desired behaviour is and if this is a MLS issue actually.

See #170 for the general issue on MCU support.

tbeu, 2017-01-19:
@sjoelund Did you had a look on this?

sjoelund, 2017-01-19:
Changing size to 0 would probably work fine (renders in almost the exact same size in OMEdit). Odd that each tool has a vastly different size for 30 pt though. SimulationX looks like it's 200~250 pt. The Real-time block is rendered decently in all tools and that uses size=0 for the font size.

tbeu, 2017-01-19:
Yes, strange indeed.

tbeu, 2017-01-19:
I believe the problem is the extend interpretation for font size > 0. It looks like its size can be neglected and is only considered for font size = 0.

sjoelund, 2017-01-19:
Yes, but that doesn't answer why the font size is so different in the different tools...

tbeu, 2017-01-19:
It is similar for Dymola and SimulationX. What is different is the vertical size of the line break.

[image]

```mo
model A
  annotation(
    Icon(
      graphics={
        Text(
          extent={{-450,450},{450,-450}},
          lineColor={28,108,200},
          fontName="Arial",
          textStyle={TextStyle.Bold},
          textString="ABCD\nABCDEF",
          fontSize=30),
        Rectangle(
          extent={{-100,100},{100,-100}},
          lineColor={28,108,200})}));
end A;
```

tbeu, 2017-01-19:
FYI: I'll file a SimulationX ticket for neglecting the font size in case of absolute font size.

dietmarw, 2017-01-20:
OMEdit:
[image]

tbeu, 2017-01-20:
MWorks:
[image]

PR #198 (2017-05-09): Use font size=0 for microcontroller
This fixes #186, where fonts are displayed in different sizes in
different GUIs when the font size is absolute.
commit fb0a0561 (2017-05-09): Use font size=0 for microcontroller

This fixes #186, where fonts are displayed in different sizes in
different GUIs when the font size is absolute.
closed 2017-05-09 by bernhard-thiele