modelica-3rdparty/Modelica_DeviceDrivers #191: What is package lmodern used for?

tbeu, 2017-01-16:
Martin, for what reason did you add `\usepackage{lmodern}` by b5713516c7c6a4468068146bd46deb8c9b2732fa for #188? In consequence, the code listings for e.g., ` GameControllerExample` or `getFoo` really have unsuitable line breaks. I'd propose to remove `lmodern` again.

See #126.

sjoelund, 2017-01-16:
It's used for the proper ℃ symbol. The regular ones look really bad.

tbeu, 2017-01-16:
The difference is marginal, but the side-effects are significant
[image] [image]
The `\degreecelsius` from `\usepackage[squaren,thinqspace]{SIunits}` ( what I usually use) looks pretty identical to default (non-lmodern) style.
[image]

tbeu, 2017-01-16:
And this is `$\SI{0}{\degreeCelsius}$` from `\usepackage{siunitx}` and `\usepackage{amsmath}`.
[image]

sjoelund, 2017-01-17:
Feel free to change it any way you want due to the side-effects. The lmodern one does look nicer than the alternatives though.

tbeu, 2017-01-17:
OK. With next PR.

commit ea5cdd9d (2017-01-18): Revert modern style

Close #191
closed 2017-01-18 by tbeu