modelica-3rdparty/Modelica_DeviceDrivers #199: Q: difference between MDD_udpSendP and MDD_udpSend

thorade, 2017-03-10:
Just a question, because from looking at MDDUDPSocket.h I was not able to figure it out:
What is the difference between MDD_udpSendP and MDD_udpSend (without P)? What should end-users use? The [Modelica wrapper](https://github.com/modelica/Modelica_DeviceDrivers/blob/master/Modelica_DeviceDrivers/Communication/UDPSocket_.mo) seems to be using udpSendP exclusively

thorade, 2017-03-10:
Nevermind, I found the difference. 
`MDD_udpSendP` expects `p_package` as input while
`MDD_udpSend` expects `data` as input.

closed 2017-03-10 by thorade

tbeu, 2017-03-10:
MDD_udpSend is the legacy API function which no longer is used in the library, but still used in the test code.

thorade, 2017-03-10:
Thanks for clarification. I will switch to using MDD_udpSendP then.