modelica-3rdparty/Modelica_DeviceDrivers #203: Incorrect Big-Endian data byte order

hyumo, 2017-03-23:
I tried to use `SerialPackager_.print()` function to print the `pkgOut.pkg` of a `AddDouble` block with big endian byte order. The printed output indicates that the `ByteOrder` was switching between big endian and little endian. 

| Environment | Version |
| ------------- | ------------- |
| Modelica | 3.2.2 |
| Dymola | 2017FD01 |
| Modelica_DeviceDrivers | 0.92.1 |
|Compiler| VS11.0, Intel 16.0 Update 3|

I tried to modify function `MDD_SerialPackagerAddDouble` within `MDDSerialPackager.h` by printing the first byte of the input value `u` before and after the byte order swap. The combined output from the Dymola message window.

> SerialPackager start: size 8, pos 8, bitOffset 0
Bytes signed dec:    d63,  d240,  d0,  d0,  d0,  d0,  d0,  d0, 
Bytes unsigned hex: 0x3F, 0xF0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 
SerialPackager end.
[Before BYTE order swap] First BYTE: 0 
[After BYTE order swap] First BYTE: 1 
SerialPackager start: size 8, pos 8, bitOffset 0
Bytes signed dec:    d0,  d0,  d0,  d0,  d0,  d0,  d240,  d63, 
Bytes unsigned hex: 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0xF0, 0x3F, 
SerialPackager end.
[Before BYTE order swap] First BYTE: 1 
[After BYTE order swap] First BYTE: 0 
SerialPackager start: size 8, pos 8, bitOffset 0
Bytes signed dec:    d63,  d240,  d0,  d0,  d0,  d0,  d0,  d0, 
Bytes unsigned hex: 0x3F, 0xF0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 
SerialPackager end.
[Before BYTE order swap] First BYTE: 0 
[After BYTE order swap] First BYTE: 1 
SerialPackager start: size 8, pos 8, bitOffset 0
Bytes signed dec:    d0,  d0,  d0,  d0,  d0,  d0,  d240,  d63, 
Bytes unsigned hex: 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0xF0, 0x3F, 
SerialPackager end.

I am a bit lost why this is happening, a demonstration of the issue with the modified C code (with in Resources\Include) could be found within the repo https://github.com/hyumo/EndianIssue. Thanks for your help.

tbeu, 2017-03-23:
It is probably not a good idea to swap the bytes on the input data of the Add-functions or the package data of the Get-functions. Can you please check if https://github.com/tbeu/Modelica_DeviceDrivers/commit/c5f19d46d98b5c520c8e2d9f27fd217361bd0871 solves your problem.

By the way, there is no v0.92.1 of M_DD. Current version is v1.4.4.

hyumo, 2017-03-24:
Thanks @tbeu , it fixes my issue (only double swap is tested, float or integer swaps are NOT) and excuse me on providing a wrong version number.

BTW, you mentioned

> It is probably not a good idea to swap the bytes on the input data of the Add-functions or the package data of the Get-functions

Is it a comment on future development or I am not using it correctly? 

As of future development, it might be good if the endianess could be specified "globally", for example, in the config block. It probably makes sense to change the endianess of a "double" if the received data is only 8 bytes (i.e. CAN messages), but if there're two "floats" stored in the 8 bytes package, then it is probably more efficient to swap order of the whole 8 bytes and do add/get functions on them. 

If I am not using it correctly, are there any suggestions on where I should change the endianess in the library? Or do I need to do something on the hardware side?

Anyways, thanks for your help.

tbeu, 2017-03-24:
Yes, it will get fixed in upcoming M_DD v1.5.0.

The reason I did not notice it earlier (when testing #66) is, that it does not occur in SimulationX (which uses copied buffers for the Add-functions) and that I did not test the Get-functions at all. Sorry for that.

Indeed, currently there is no model-wide setting (like e.g., Modelica.Mechanics.MultiBody.World) for the endianess configuration. I have no idea if we should go this direction.

closed 2018-02-01 by tbeu