modelica-3rdparty/Modelica_DeviceDrivers #209: Release v1.5.0

bernhard-thiele, 2017-05-09:
@tbeu I have drafted a release for v1.5.0. which I would like to release this Friday. I will do a bit more checking with Dymola and OpenModelica. It would be great if you had some time for checking with SimulationX, too. There are no new changes except for various documentation updates, so I hope that there are no new issues.

tbeu, 2017-05-09:
Ordered by ticket number:

- Are you going to merge #135 for M_DD v1.5.0? Once, I know your decision, I'll provide new binaries by a PR.

- I am not sure if emphasizing the importance (and backward incompatibility) of #138 is really necessary in the release notes, since nobody seemed to use it anyway.

- I am also in favour of fixing #184. It is a tiny fix only.

- Arr you going to merge #201 for M_DD v1.5.0?

- I'd also explicitly list #203/#206 in the release notes, since this is a rather relevant bug fix (triggered by user).

tbeu, 2017-05-09:
I've just contacted Travis CI support about the Precise configuration error. I'll let you know once I know more.

bernhard-thiele, 2017-05-10:
- I don't plan to merge #135.
- For #138 I'm not so sure that nobody used it. If so, they are at least warned, if nobody used it, no harm is done by mentioning it.
- Ok, let's fix #184. I suppose I should fix the Linux side, while you take care for Windows.
- I played with #201 but simple merging will add many commits that are not related to the source code and clutter the version history. Also, the paper is something fixed (we won't update it like the rest of the library), so shipping its sources with the library doesn't feel fitting so well anyway. I now lean more towards leaving it as its own side project in its branch and keep the branch. The paper is also not yet officially published, it is a (final) draft version. The official paper will go through some processing by editors (adding a doi etc.). It seems more practical to simply add a link from the documentation to the paper as soon as it is published an publicly available.
- I will add a note for #203 to the release notes.

bernhard-thiele, 2017-05-10:
I just checked for lcm. The latest release is v1.3.1 (https://github.com/lcm-proj/lcm/releases). That is the same release that I use in the cmake build scripts. Are you proposing to use a "master" snapshot, or did we build on different library versions in Linux and Windows?

tbeu, 2017-05-10:
> Are you proposing to use a "master" snapshot ...?

No, I propose to use the the v1.3-lts branch of lcm, which compared to the v1.3.1 release contains the fix for the resource leak. I'll provide PRs soon.

bernhard-thiele, 2017-05-12:
Released.

closed 2017-05-12 by bernhard-thiele