modelica-3rdparty/Modelica_DeviceDrivers #215: Scalable real-time synchronization

bernhard-thiele, 2017-05-25:
Sometimes it is useful to run faster or slower than real-time. Examples are human-in-the-loop scenarios which have phases in which nothing "interesting" happens in which case one would like to fast-forward the simulation, e.g., in flight simulations during an autopilot phase.

For supporting such use-cases the `realtimeSynchronize` should be extended with an (optional) real-time scaling input signal.

tbeu, 2017-05-26:
@nytschgeusen That is the issue we were talking about.

bernhard-thiele, 2017-05-26:
@tbeu Ok, I also talked with @nytschgeusen during the Modelica conference and got convinced that it is a useful feature to go for :-)

I hope @nytschgeusen can do some testing and give feedback if I got the feature right.

I had to change the C code, so that the SimulationX DLLs need to be recompiled. 

By the way, maybe it makes sense to change to https://git-lfs.github.com/ at some point for storing the binaries. The service sounds interesting, but I have so far no experience with it.

closed 2018-07-31 by bernhard-thiele