modelica-3rdparty/Modelica_DeviceDrivers #224: Make Release v1.5.1

beutlich, 2017-08-23:
Because of 59520564f81dad78bebd9e1e7230f68f1b2a0fff I still recommend [v1.4.4](https://github.com/modelica/Modelica_DeviceDrivers/releases/tag/v1.4.4) rather than latest [v1.5.0](https://github.com/modelica/Modelica_DeviceDrivers/releases/tag/v1.5.0) to our users. Do you think you can create a patched release v1.5.1 with 59520564f81dad78bebd9e1e7230f68f1b2a0fff applied on bb2b854a5971e993ee244375d8c6f3ae43e79722?

bernhard-thiele, 2017-08-24:
Yes, makes sense to have bugfix release

bernhard-thiele, 2017-08-24:
I have now a [draft release](https://github.com/modelica/Modelica_DeviceDrivers/releases/tag/untagged-9cd33feb16f78094a64b). Maybe you can have a quick look at it. Should be fine, but cannot test it in the moment. Once a bit of sanity testing has been done we can release it.

tbeu, 2017-08-28:
Thanks. I compared Modelica_DeviceDrivers_v1_5_1.zip and Modelica_DeviceDrivers_v1_5_0.zip and could see additional differences in EOL style. Furthermore the version annotation in package.mo needs to be incremented.

bernhard-thiele, 2017-09-04:
Thanks, I fixed the version annotation.
Also, I looked at the EOL style in Modelica_DeviceDrivers_v1_5_0.zip and the few files that I checked had Unix EOL style. Comparing with the latest uploaded Modelica_DeviceDrivers_v1_5_1.zip I didn't see any additional differences (to this latest version).

bernhard-thiele, 2017-09-19:
@beutlich If you have no more comments I would release the draft today in the evening.

bernhard-thiele, 2017-09-19:
Released.

closed 2017-09-19 by bernhard-thiele