modelica-3rdparty/Modelica_DeviceDrivers #226: Maintenance of ClockedBlocks?

tbeu, 2017-09-19:
Currently, compared to the (continuous) Blocks the ClockedBlocks are not well maintained and do neither work in OpenModelica nor SimulationX. Instead of putting a lot a maintenance work in to the library to improve the tool support and Modelica compliance of the ClockecBlocks, I wonder if these blocks are really necessary at all. Since the external trigger enhancement of the (continuous) Blocks in #77 and together with the clock2Trigger block (=Modelica_Synchronous.RealSignals.Sampler.Utilities.AssignClockToTriggerHold) it seems to me the ClockedBlocks are obsolete.

This is an example how the (continous) Blocks are triggered by synchronous input.
[image]

@bernhard-thiele If you agree, my proposal is to declare the ClockedBlocks as obsolete (using the obsolete annotation and the inheritance from Modelica.Icons.ObsoleteModel).

@FrenkelJ FYI

bernhard-thiele, 2017-09-19:
I don't like to make them obsolete, I would rather like to improve them where necessary. The synchronous extension was added to Modelica with the intention to improve discrete-time modelling for sampled data systems and this fits very well with the MDD library.

Particularly, since we now have more tool support for synchronous, it gets more attractive to put more efforts into these blocks. If it should turn out, that it is difficult to realize standard conform synchronous blocks which support the functionality needed in MDD, this would be an input to the Modelica standardization that should be interesting for the language design.

tbeu, 2017-09-19:
I see your point. My intention rather was to have a short-term solution, given the time we spent on the (continuous) Blocks, the expected work on the maintenance of the ClockedBlocks and the resources we have. Still, do you think the ClockedBlocks can be substitued by (continuous) Blocks with external trigger and the clock2Trigger block?

bernhard-thiele, 2017-09-19:
The workaround above is possible, but if you instead receive data there might be a subtle problem. The sample operator introduces an infinitesimal small delay (Section 16.5.1). Hence, if you have the `clockToTrigger` block, which triggers a reading operation whose result is then fed into the sample operator for using it in some clocked processing logic, and the _same_ clock triggers the subsequent processing logic, the actual reading might be executed after the processing logic (instead of before, which is probably intended).

MartinOtter, 2017-10-22:
Clocked partitions are the preferred way to model sampled-data systems in Modelica (due to the many benefits). At some point in the future, sampled-data systems should be only described with clocked partitions and no longer by emulations with continuous-time systems. Therefore, it is the other way round: The continuous-blocks in Modelica_DeviceDrivers have to be marked "obsolete" at some point in the future.
Please, can you explain why ClockedBlocks do not work in SimulationX and in OpenModelica (maybe this is just a minor issue that needs to be fixed in these tools).

tbeu, 2017-10-22:
> Please, can you explain why ClockedBlocks do not work

It all is a library issue. The Modelica_DeviceDrivers library was developed with only Dymola supporting it. In 2015, we spent quiet some time to get Modelica_DeviceDrivers.Blocks and Modelica_DeviceDrivers.Packaging.SerialPackager somehow Modelica conform - with the exceptions from the MLS mentioned in the 2017 [article](http://dx.doi.org/10.3384/ecp17132713), section 3. This was the fundamental requirement that other tools like SimulationX and OpenModelica are able to support Modelica_DeviceDrivers.Blocks. At that time (2015), no other tool than Dymola supported the Modelica Synchronous features (of MLS chapter 16) and therefore, the Modelica_DeviceDrivers.ClockedBlocks are not yet Modelica conform to be supported in other tools than Dymola. This again will require some resources - cannot estimate how large the effort will be.

I agree with you, that Modelica_DeviceDrivers.ClockedBlocks are the future proof way for sampled device I/O and therefore should work the same way as th continouos Modelica_DeviceDrivers.Blocks. Just skip the idea with clock2Trigger block (=Modelica_Synchronous.RealSignals.Sampler.Utilities.AssignClockToTriggerHold), which I considered a (nice and working) short-term solution. Thank you for that block, which was the reason why I raised this issue when I discoverd it.

tbeu, 2017-11-09:
OK, no more open questions.

closed 2017-11-09 by tbeu