modelica-3rdparty/Modelica_DeviceDrivers #233: UDP communication decode

MustafaSoliman, 2018-03-20:
I am trying to use Modelica_DeviceDrivers.Blocks.Communication.UDPSend
 function  to send data to another program via UDP communication but the data encryption is not mentioned in the documentation or file. 
the data is sent but i can't decode it. 
any help?

MustafaSoliman, 2018-03-20:
i tried too many protocols ascii, UTF-8, gpk but it didn't work. and i have the messages like the files below. 
[log1.txt](https://github.com/modelica/Modelica_DeviceDrivers/files/1827662/log1.txt)
[log2.txt](https://github.com/modelica/Modelica_DeviceDrivers/files/1827663/log2.txt)

bernhard-thiele, 2018-03-20:
Section 2.5.2 in http://www.ep.liu.se/ecp/132/078/ecp17132713.pdf describes the format. You can also have a look at the test examples for the serial packager in subfolder Modelica_DeviceDrivers/Resources/test/Communication

closed 2018-03-20 by bernhard-thiele