modelica-3rdparty/Modelica_DeviceDrivers #236: Provide number of received bytes in UDPReceive block as output

bernhard-thiele, 2018-07-31:
Sometimes one would like to know how many new bytes were received. Particularly, one would like to know if a new package has been received at all (hence, received bytes > 0).

The idea is to introduce an additional integer output `nRecvBytes` to the UDPReceive blocks. This output will be `0` if no new package was received, otherwise its value will be the number of received bytes.

bernhard-thiele, 2018-07-31:
Additionally, there could be an output `nRecvbufOverwrites` which allows to check whether the receive buffer was overwritten with new data before the last data was retrieved by the Modelica simulation.

tbeu, 2018-08-05:
Do you want me to update the SimulationX-specific ITI_MDD.dll already now?

bernhard-thiele, 2018-08-06:
Yes that would be fine, since I don't have another PR in the pipeline. Could also be good idea to integrate a SimulationX target in the CMakeLists.txt file so that a user of the master branch can easily recompile for SimulationX, if needed.

tbeu, 2018-08-07:
> Could also be good idea to integrate a SimulationX target in the CMakeLists.txt file

We have it already, don't we?

tbeu, 2018-08-24:
The commited changes actually break the API for the UPD send functions, both on Modelica and C level. Should we rather add (and utilize) new functions and keep the existing functions (as obsolete), in order to not break custom models relying on the function layer?

bernhard-thiele, 2018-08-24:
I don't think this will be an issue since existing code should still work. Since there had been no return value, existing code will not check for any return value, hence it should be backwards compatible, shouldn't it?

tbeu, 2018-08-24:
It can be an issue for copied code (for example, https://github.com/RWTH-EBC/AixLib/blob/a4b2a05dcf0f6db46341851ef3ea4957cfcb2e8c/AixLib/Controls/HVACAgentBasedControl/BaseClasses/Internal/sendTo.mo)

tbeu, 2018-08-28:
To be on the safe side and in compliance with semantic versioning I'd recommend to introduce new/extended functions or to label the new version 2.0.0.

bernhard-thiele, 2018-08-29:
Hm, in general I don't see the signature of external C functions as part of the public API of a Modelica library, but rather as a part of the internal implementation. However, it's not a big deal to add some functions to maintain bachwards compatibility of the C layer in this case, so I will do it.

commit 34a538e7 (2018-08-01): UDPReceive blocks enhancements. Closes #236

1. Additional output nRecvBytes. Output is not visible on diagram layer
as default, but can be made visible by a parameter in the "Advanced" tab.
2. Additional output nRecvbufOverwrites.
commit f7996e0c (2018-08-01): Merge pull request #237 from bernhard-thiele/bth-udpext

UDPReceive blocks enhancements. Closes #236
commit c43f5ae1 (2018-08-06): close #236: Update SimulationX specific binaries
commit fe366bca (2018-08-29): Backwards compatible `MDD_udpReadP(..)`. Closes #236
closed 2018-08-29 by bernhard-thiele