modelica-3rdparty/Modelica_DeviceDrivers #273: Allocation of additional '\0' byte needed when using ModelicaAllocateString(len)!?

bernhard-thiele, 2019-01-11:
So far I assumed that `ModelicaAllocateString(len)` would automatically allocate an additional '\0' byte after the end of the requested string length.

The OpenModelica implementation does indeed add a '\0':

https://github.com/OpenModelica/OMCompiler/blob/da971419804d708e27b866e4f017d1543ab7a0c7/SimulationRuntime/c/util/ModelicaUtilities.c#L89-L95

However, with Dymola (using version 2019 FD01) it seems necessary to manually allocate the extra `\0` (at least it seemed to fix a bug for me in some code). Looking into the Modelica Specification v3.4 on p. 174 it is not clear to me whether argument `len` is supposed to be the `string_length` or `string_length + 1`?

The safest fix is probably to include the extra byte in the MDD code. Probably it doesn't matter if there are internally two successive '\0' characters in some tools, but it matters if there is no '\0'. Therefore, include a '+1' in code snippets like:

https://github.com/modelica/Modelica_DeviceDrivers/blob/6c789ddec0adaedf910019fff39055d9c7551e72/Modelica_DeviceDrivers/Resources/Include/MDDSerialPackager.h#L487

tbeu, 2019-01-11:
Hm, I doubt that Dymola changed the implementation. You can see the `(len+1)` in the sources of `StringAllocate`:

```
DYMOLA_STATIC char* StringAllocate(SizeType len) {
	char*ret=0;
	EnsureMarkInitialized();
	ret=(char*)stringmark;
	Assert(stringmark+len+1<Endsimplestring,"Room to allocate string");
	stringmark+=(len+1);
	return ret;
}

DYMOLA_STATIC char* ModelicaAllocateString(size_t len) {
	char*s=StringAllocate(len);
	s[0]='\0';
	return s;
}
```

Furthermore, in MSL there are many calls of `ModelicaAllocateString(strlen("someString"))` (e.g., https://github.com/modelica/ModelicaStandardLibrary/blob/d12a25282a64d5b5b7e4cc526b620aabe705b08f/Modelica/Resources/C-Sources/ModelicaInternal.c#L1057) which would fail otherwise.

sjoelund, 2019-01-12:
The bugs seems to be in MDD, but is one of the things OpenModelica protects against. Dymola seems to add a `\0` to the beginning of the string whereas OM does it at the end. The `memcpy` used does not add a trailing 0-byte as required. It checks that following parts of the buffer contain `\0` but not in the current segment. At least something like that is what I get from reading the linked code.

bernhard-thiele, 2019-01-14:
Thanks for the clarifications! 

So Dymola allocates the additional space for `\0`, but requires the user to ensure that it is provided. I assume also SimulationX is allocating an additional space for `\0`, so that it is not necassary to ask for a `len` which is `string_length + 1`.

I will fix that in MDD. Maybe a clarification about ModelicaAllocateString(len) in the specification wouldn't do any harm...

tbeu, 2019-01-14:
> Maybe a clarification about ModelicaAllocateString(len) in the specification wouldn't do any harm...

The name of the argument is `len` (different from `size`, which is `len+1`) makes it clear that the string length shall be passed, but I agree that should be more obvious. Will you create a PR?

bernhard-thiele, 2019-01-14:
OK, I will propose a PR.

commit ff5d6838 (2019-01-14): Fix missing byte copy of '\0' in MDDEXT_SerialPackagerGetString()

Fixes #273
closed 2019-01-14 by bernhard-thiele