modelica-3rdparty/Modelica_DeviceDrivers #275: Support blocking UDP receive call

bernhard-thiele, 2019-02-01:
Rationale: Support of a blocking UDP receive call allows synchronizing a simulation to an external application sending the UDP datagrams.

The proposal is to add an option `useRecvThread` to the UDPReceive block. So far a separate thread is created for receiving datagrams (corresponds to the proposed default value `useRecvThread=true`). By setting the option to `false` a user can prevent that a separate thread is created and simulation progress is blocked when calling out to UDP `recvfrom(..)`.

[image]

bernhard-thiele, 2019-02-01:
Yes, I'm not sure what the better word is. But thinking about it, I lean to "dedicated". I will change it.

beutlich, 2019-02-01:
> I will change it.

Thanks.

And please also add a separate test for it.

closed 2019-02-12 by bernhard-thiele

bernhard-thiele, 2019-02-12:
I've now  also successfully tested the code using OpenModelica running on Linux. The code seems okay and thus I squashed & merged it.

PR #276 (2019-02-12): Blocking UDP feature
Closes  #275
commit 84b37bd8 (2019-02-12): Blocking UDP feature (#276)

Support blocking UDP receive call (closes #275)