modelica-3rdparty/Modelica_DeviceDrivers #281: Invalid Value "showReceivedBytesPort" for Placement.visible

tbeu, 2019-03-15:
SimulationX reports
```
Warning in Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_UDP.uDPReceive.nRecvBytes.Placement.visible

Invalid Value "showReceivedBytesPort" for "Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_UDP.uDPReceive.nRecvBytes.Placement.visible".
Help: Modelica.BadValueWarn
```
Should DynamicSelect be used in https://github.com/modelica/Modelica_DeviceDrivers/blob/d4d9a2f6fbc2985431222a6b08c6f105223fd3b6/Modelica_DeviceDrivers/Blocks/Communication.mo#L122-L123 and https://github.com/modelica/Modelica_DeviceDrivers/blob/d4d9a2f6fbc2985431222a6b08c6f105223fd3b6/Modelica_DeviceDrivers/ClockedBlocks/Communication.mo#L145-L146 ?

bernhard-thiele, 2019-03-18:
Hm, as I understand `DynamicSelect` it is for animations, but here I only want to control the visibility of the annotation during editing. One could do

`visible = DynamicSelect(showReceivedBytesPort,showReceivedBytesPort)`

which also gives the desired behavior during editing in Dymola. Would that work correctly in SimulationX?

Still, it would be interesting to understand whether simply using `visible=showReceivedBytesPort` is correct from an MLS point of view...

tbeu, 2019-03-18:
Ah, sorry. This is a tool issue.

closed 2019-03-18 by tbeu